python-microscopy/python-microscopy #581: [ENHANCEMENT] concatenate tiled series

barentine, 2020-11-04:
**Is your feature request related to a problem? Please describe.**
I have a lot of tiled overview images that I want to view as one supertile source (just for viewing at the moment)

** *Potential soutions* **

**pyme-cluster:// - aggregate ImageOutput module:**
Add to HTTPDataServer aggregate_h5 endpoint such that we could enable `pyme-cluster:// - aggregate` scheme for `output.ImageOutput`. This would not entail much in the dataserver, but rather in `IO.h5File.H5File`, where `put_file` would need to have a bit more logic on pzf frames (since frame numbers will overlap).

However, because `PointScanner` events, `ScannerXPos, ScannerYPos` only include time stamps and not frame numbers, and we use `mdh['StartTime']` to map to frames, we would need to spoof the event times when appending them and associated frames (would also need a lock taken out in part of `H5File.put_file` to update a max_frame / max_time variable in case we try and push frames from multiple sources at once. This might be easiest in a separate method to `put_file` that can take events, frames, and metadata all at the same time.

**SuperClusterPZFDataSource:**
A datasource for the superlative extraordinaire who wants to make a SupertileDataSource from only the finest input datasource. Would be grosser from a ui perspective in my opinion, as the other I can make happen behind the scenes easily in a recipe I'm already running for detection/queueing on individual tile overviews, while this would I would somehow need to generate an arbitrarily long list of inputs for a recipe module.

**Are you prepared to work on it?**
always

barentine, 2020-11-04:
modifying the tileviewer might be super simple from the openseadragon perspective as we just need to add more tile sources. 

adding a `multitileviewer.py` and associated `multitileviewer.html` really might be straightforward

David-Baddeley, 2020-11-04:
I think any aggregation of separate tile pyramids is going to be a bit tricky due to the fact that it's highly unlikely that the tile origins will align on the same grid.  The 3 options I'd consider would be:

a) save everything to the one pyramid (this could be done from separate, per-well protocols, but ultimately all writing into the same pyramid structure). This would ensure tiles are aligned and make viewing a cinch. Might not be practically realisable until we have a distributed version of the tile pyramid though.

b) do something on the viewer end to load and display separate tile pyramids in the one view. May not be super simple due to the potential for tile boundary misalignment.

c) if just for an overview for publication / presentations, pull the top level tiles from each pyramid and write them into an image in a custom script / ipython notebook

David-Baddeley, 2020-11-04:
e) display N separate pyramids, one for each well (or display a non-zoomable overview as in c, and have clicking on a specific well open that well in the openseadragon viewer

barentine, 2020-11-05:
After discussion with David:
Only need at the moment is for display in presentations / publication. Will write a jupyter notebook to load a list of pyramids as top-layer supertiles and generate the super supertile image we're looking for

barentine, 2020-11-18:
now taken care of in an example notebook

closed 2020-11-18 by barentine