python-microscopy/python-microscopy #582: [SUPPORT] state handler for cameras?

barentine, 2020-11-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when i have to stop frameWrangler, change a bunch of stuff, prepare and restart it in the top of every protocol when the logic could be as simple as `scope.state.update`

**Describe the solution you'd like**
`microscope.register_camera` could add state handlers for integration time, multiview active views, multiview roi size, etc.

**Describe alternatives you've considered**
bunch of manual stuff in everyone's protocols

**Are you prepared to work on it?**
yes

barentine, 2020-11-04:
Note - I see handlers added in microscope init for the camera:
- Camera.IntegrationTime
- Camera.ROI
- Camera.Binning

I guess that just leaves the multiview ones, which I can add a method for to the multiviewcameramixin

David-Baddeley, 2020-11-04:
As noted, it's mostly already there. 

For a bit of context, the handlers are in the microscope object rather than being deferred to the camera (which might initially seem more intuitive) to facilitate multi-camera support. `scope['Camera.IntegrationTime']` should set the integration time of the active camera, and ultimately it's the scope object that knows which camera is active.

David-Baddeley, 2020-11-04:
As a side note, setting any of these in protocols is very risky - the metadata and most of the file backends assume a constant ROI size, pixel size, and integration time for the entire series.

barentine, 2020-11-04:
Changing the active views in the protocol works for pzf spooling, which is great, and what I use in `htsms-tile.py`, but you're right that the start metadata is wrong. Any suggestions for how to fix that? I'm assuming swapping Protocol.Init and doStartLog in `StartSpool` is out of the question.

Was actually about to take a two-color tile series today where I'll need the metadata right later to split it - thanks for pointing this out.

barentine, 2020-11-04:
Alternatively could define a `protocol.set_state` method to call before `doStartLog`, and let people define an `INITIAL_STATE` dict in the protocol, but I'm not actually seeing the problem with swapping doStartLog and protocol.Init

barentine, 2020-11-05:
Multiview state handlers queued to be added in #578. 
Swap to log start metadata after frame '-1' tasks queued in #582.

barentine, 2020-11-18:
fixed with #583 and #578

closed 2020-11-18 by barentine