python-microscopy/python-microscopy #584: [ENHANCEMENT] Make it easier to visualize and export measurements from PYMEVis

zacsimile, 2020-11-05:
**Is your feature request related to a problem? Please describe.**
Measurements, particularly cluster measurements, can be difficult to use/plot from within PYME. It is often easier for users to export this data to CSV. The menu item `File > Save measurements` offers hope where none is to be found. To export, one needs to type `pipeline.selectedDataSource.toDataFrame().to_csv('my_csv_name.csv')` into the shell. 

**Describe the solution you'd like**
- Make `File > Save measurements` not dependent on `objectMeasures`
- Make it easy to export data to CSV (`File > export`)
- Make it easy to plot two variables from a data source against one another, either as a popup or in a new tab on PYMEVis (`View > Plot variables`)

**Are you prepared to work on it?**
Yes

David-Baddeley, 2020-11-05:
Probably best handled by reinstating the array viewer.

David-Baddeley, 2020-11-05:
PYME.ui.recArrayView as base

David-Baddeley, 2020-11-22:
Going to merge #624 in the interim, and put integrating with #594 on the todo list. Will fix some of the making sure it displays on top of #633.

Wishlist for future work (copied in part from comments on #624, some of this already in #594):

- sorting
- vertical recipe integration
- make sure this displays for measurements
- graphing
- summary statistics?
- to excel / to jupyter buttons??? (single click to open a notebook pre-populated with code to load this data)???
- fitting?? Integrate with e.g. line profile stuff?