python-microscopy/python-microscopy #592: [SUPPORT] Frequency of package bulids

zacsimile, 2020-11-06:
It would be nice to build the `pip`/`conda` packaged versions of PYME with greater frequency. We have made a lot of changes since September that make PYME much more user-friendly, but ironically the only way to access these is through a developer install. We have a package workflow that can in theory be automatically triggered. Can we automatically build a package once per week? Or at least build a package every time `PYME.version.version` changes?

David-Baddeley, 2020-11-06:
Automatic building is on the medium term TODO list, but I don't feel we're quite ready yet (w.r.t test coverage,  test pass rate, etc ...). For now, manually triggered builds have a few advantages (we know roughly what state things were in for the last build). I've been meaning to trigger a new build for a few weeks now, but 
a) there's been a lot of activity (and associated regressions) so I was waiting for those to shake out a little and 
b) there are a couple of things on the TODO list that I'd like to get done first. The blocking items are as follows:

- work out the pytables/blosc version issues and update pymedepends pins accordingly
- ditto for traitsui (less of an issue than above as only effects py27, probably not actually blocking)
- update the minimum pyme-depends version in the main package
- update the install docs now that the default miniconda version is 3.8 (need to downgrade python to 3.7 before installing pyme)

zacsimile, 2020-11-06:
I agree with you that we are not quite ready, and I'd like to make moving toward increasing the build frequency a priority. The biologists here often want new features/slight tweaks and fixes to existing features. As such, because of the delay between releases, I recommend they use a development install. I think it would be better for them to use the single-click installer, but I can't push fixes to their copies of PYME at the rate they want unless we increase build frequency.

PYME is constantly in flux and it seems likely this will not change in the near future. As such, it may not be prudent to wait for all developments to settle before releasing new builds. What do you think is a reasonable release schedule?

Rather than updating the install docs, is now a good time to start support Py 3.8? What are the bottlenecks there?

We are definitely low on test coverage, but our pass rate looks like it's <100% only due to skipped tests on the most recent PR (#591)! 

[image]

Should we start uncommenting these tests one by one, and fixing them when necessary, to improve pass rate/coverage?

zacsimile, 2021-03-16:
I think everything on your blocking list is done, @David-Baddeley. Granted our test coverage still isn't great, but should we move to automatic builds?

David-Baddeley, 2021-03-16:
I'd like to keep the executable installers such that they are (at least semi) manually audited. I also kind of like the idea of having manual releases, where we can make sure that any pressing bugfixes are merged etc ... before releasing.

Fully automated builds should probably go into a different channel.

Steps towards full automation:

- combine the 3 manual steps of the current build pipeline so that you only need to trigger a release in one place.
- find a branch protection workaround for the release action
- better test coverage (I'd be aiming for somewhere north of 70% before wanting to commit to fully automated builds, although we could possibly move earlier if they were going to a separate channel)

I'll note that our release frequency is now much better than when this issue was raised.