python-microscopy/python-microscopy #598: [BUG] enum / File types tricky with html recipe editor

barentine, 2020-11-09:
**Describe the bug**
running the tornado clusterUI, on recipe launch
```
ERROR:jigna.server:Traceback (most recent call last):
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/jigna/server.py", line 93, in handle_request
    result = method(request)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/jigna/server.py", line 205, in set_instance_attribute
    setattr(obj, attribute_name, value)
  File "/home/smeagol/code/traits/traits/base_trait_handler.py", line 77, in error
    object, name, self.full_info(object, name, value), value
traits.trait_errors.TraitError: The 'scaling' trait of a RGBImageUpload instance must be 'min-max' or 'percentile', but a value of 'min-ma' <class 'str'> was specified.
Traceback (most recent call last):
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/jigna/server.py", line 93, in handle_request
    result = method(request)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/jigna/server.py", line 205, in set_instance_attribute
    setattr(obj, attribute_name, value)
  File "/home/smeagol/code/traits/traits/base_trait_handler.py", line 77, in error
    object, name, self.full_info(object, name, value), value
traits.trait_errors.TraitError: The 'scaling' trait of a RGBImageUpload instance must be 'min-max' or 'percentile', but a value of 'min-ma' <class 'str'> was specified.

```

see discussion in #534

barentine, 2020-11-09:
too quick on the trigger - different issue on launch. Sorry

closed 2020-11-09 by barentine

David-Baddeley, 2020-11-09:
Still a known problem with the recipe editor as it stands (the chance or reliably getting the enum string right is pretty low), but probably doesn't strictly need it's own issue.