python-microscopy/python-microscopy #599: [BUG] recipe rules using inputsByTask get all tasks assigned to a single node

barentine, 2020-11-09:
**Describe the bug**
launching a recipe from the cluster over multiple series, using e.g. `run` view, results in all (in this case 22 tasks) getting assigned to a single node

**To Reproduce**
Steps to reproduce the behavior:
1. run either clusterUI backend
2. navigate to the run view
3. glob multiple series, select a recipe, launch
4. watch them all get assigned to the same node

**Expected behavior**
bidding on tasks to actually bid on tasks rather than rules

**Additional context**
failed recipe, but
<details> 
```
===========================================================================
Error in rule: 000014-4b2f026e1cc846aaab3988a3d26e03ff running task: 21 on drno:19399

taskDescr:
----------
{'id': '000014-4b2f026e1cc846aaab3988a3d26e03ff~21', 'type': 'recipe', 'inputs': {'input': 'pyme-cluster:///Bergamot/2020_10_27/EMC07_02_S1/A1/000/analysis/27_10_series_T.h5r'}, 'taskdefRef': 'pyme-cluster://villains/RECIPES/emc0702s1-map-and-omero.yaml'}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 340, in computeLoop
    recipe.execute()
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 640, in execute
    m.check_inputs(self.namespace)
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 137, in check_inputs
    raise MissingInputError('Input "%s" is missing from namespace; keys: %s' % (input, keys))
PYME.recipes.base.MissingInputError: Input "Events" is missing from namespace; keys: ['FitResults']

===========================================================================
Error in rule: 000014-4b2f026e1cc846aaab3988a3d26e03ff running task: 20 on drno:19399

taskDescr:
----------
{'id': '000014-4b2f026e1cc846aaab3988a3d26e03ff~20', 'type': 'recipe', 'inputs': {'input': 'pyme-cluster:///Bergamot/2020_10_27/EMC07_02_S1/A1/000/analysis/27_10_series_M.h5r'}, 'taskdefRef': 'pyme-cluster://villains/RECIPES/emc0702s1-map-and-omero.yaml'}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 340, in computeLoop
    recipe.execute()
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 640, in execute
    m.check_inputs(self.namespace)
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 137, in check_inputs
    raise MissingInputError('Input "%s" is missing from namespace; keys: %s' % (input, keys))
PYME.recipes.base.MissingInputError: Input "Events" is missing from namespace; keys: []

===========================================================================
Error in rule: 000014-4b2f026e1cc846aaab3988a3d26e03ff running task: 19 on drno:19399

taskDescr:
----------
{'id': '000014-4b2f026e1cc846aaab3988a3d26e03ff~19', 'type': 'recipe', 'inputs': {'input': 'pyme-cluster:///Bergamot/2020_10_27/EMC07_02_S1/A1/000/analysis/27_10_series_L_1.h5r'}, 'taskdefRef': 'pyme-cluster://villains/RECIPES/emc0702s1-map-and-omero.yaml'}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 340, in computeLoop
    recipe.execute()
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 640, in execute
    m.check_inputs(self.namespace)
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 137, in check_inputs
    raise MissingInputError('Input "%s" is missing from namespace; keys: %s' % (input, keys))
PYME.recipes.base.MissingInputError: Input "Events" is missing from namespace; keys: ['FitResults']
[... 302 lines cut ...]
    m.check_inputs(self.namespace)
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 137, in check_inputs
    raise MissingInputError('Input "%s" is missing from namespace; keys: %s' % (input, keys))
PYME.recipes.base.MissingInputError: Input "Events" is missing from namespace; keys: ['FitResults']

===========================================================================
Error in rule: 000014-4b2f026e1cc846aaab3988a3d26e03ff running task: 1 on drno:19399

taskDescr:
----------
{'id': '000014-4b2f026e1cc846aaab3988a3d26e03ff~1', 'type': 'recipe', 'inputs': {'input': 'pyme-cluster:///Bergamot/2020_10_27/EMC07_02_S1/A1/000/analysis/27_10_series_R_1.h5r'}, 'taskdefRef': 'pyme-cluster://villains/RECIPES/emc0702s1-map-and-omero.yaml'}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 340, in computeLoop
    recipe.execute()
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 640, in execute
    m.check_inputs(self.namespace)
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 137, in check_inputs
    raise MissingInputError('Input "%s" is missing from namespace; keys: %s' % (input, keys))
PYME.recipes.base.MissingInputError: Input "Events" is missing from namespace; keys: ['FitResults']

```
</details>

David-Baddeley, 2020-11-09:
I think this is probably down to the default chunk size and the first-come, first-served task allocation. IE nodeserver will bid correctly on tasks, but will do so on `50*num_processors` tasks (see `Nodeserver.num_tasks_to_request`). Because the ruleserver assigns tasks on a first come, first served basis, the first nodeserver to get a bid in gets all the tasks (and then passes them on to it's first worker).

There are a couple of ways to fix this:

a) make the task allocation logic in the ruleserver a bit more sophisticated - i.e. actually look at the costs before assigning tasks rather than just assuming that a worker will bid preferentially on local tasks. As it stands the ruleserver trusts the nodeservers to bid  preferentially on tasks where the data is local. This works really well for localisation tasks, but due to the chunk size, is not great for recipe tasks.
b) improve the nodeserver logic so that it only asks for one recipe task at a time (rather than 50*num_processors), or potentially doesn't bid on non-local recipe tasks.
c) some combination of both.

I'm pretty sure I added a property to the rule description to let you set the ideal chunk size for that rule, but I don't think I wired anything up on the backend and am not sure where that property is defined.

David-Baddeley, 2020-11-10:
The simplest and least controversial change would be to change line 227 in rulenodeserver.py from 

`if n_tasks < n_tasks_to_request:` to `if n_tasks == 0:`

This would mean that the nodeserver would not bid on any non-local tasks unless it couldn't find any local ones (as it stands, it will bid on non-local any time it doesn't get 50*num_procs tasks).  There are still a few edge cases here (data distributed unevenly on cluster and a server doesn't have any local tasks) which could trigger gross behaviour, but this would already represent quite an improvement. It would also make one of the possible changes to the ruleserver more palatable.

On the ruleserver end a reasonable change would be to do a basic cost check on the and return straight away if the cost was below a certain threshold (i.e. local), otherwise stake a claim to the tasks and return after, a certain time (e.g. 2-300 ms) if no-one had bettered the claim.  I've got a pretty clear idea of what this should look like, so might try and quickly code up an initial punt.

David-Baddeley, 2020-11-10:
See #601 for a rough implementation

David-Baddeley, 2020-11-10:
Note, even with #601, we might still end up sending all tasks to a single worker on each node.

David-Baddeley, 2020-11-10:
Actually the above should be solved by specifying `optimal-chunck-size=1` in the task template (see rulenodeserver line 366)

PR #1088 (2021-07-12): default chunk size for recipes to 1
Final bit in preventing all recipes from landing on a single node, closes #599
closed 2021-07-12 by David-Baddeley