python-microscopy/python-microscopy #603: [BUG] Data Pipeline view destroyed by hitting "Apply Text Changes" after adding an OutputModule

zacsimile, 2020-11-10:
**Describe the bug**
This is a weird and minor bug, but possible to hit if you're trying to build a pipeline. Suppose you've build a pipeline and you want to save results. 

[image]

You think you're clever with the recipe builder, so you add an output module.

[image]

It appeared in the pipeline! Great, you think, this must work. But you go to your save directory and nothing is there. So, you hit `Apply Text Changes` to force the recipe to re-run.

[image]

Suddenly, the `Data pipeline` is messed up and still nothing has saved. There was no error caught by the context manager. What's going on? You go to the command line and all it says is the following.

```
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/vertical_recipe_display.py", line 274, in OnPaint
    x0, y0 = self._input_position(nd)
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/vertical_recipe_display.py", line 220, in _input_position
    xp, yp = item.Position
```

**Expected behavior**
Probably two things.

1. Don't let a user add an output module (re: #440, unless we've changed our philosophy).
2. Rebuild the pipeline display correctly, even if it's a non-functioning pipeline, or throw an error telling the user it can't be rebuilt because module X makes this a broken pipeline.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15.7
 - Python version: 3.6.10
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2020-11-10:
could you provide the full traceback (i.e. what error was thrown)?

I suspect this is could be result of something going wrong with the input of the module, but not sure what. It's super, super odd that the vertical display show initially but then doesn't rebuild when you apply text changes (by the way, 'apply text changes' is a lousy way to trigger a re-run, and probably won't do so in the future, especially if the recipe is unchanged). A number of things we should do ...

- make it clear in the output module docs that these are ignored unless doing batch processing
- potentially adjust the display to also indicate this (e.g. not show output modules in pipeline view and show them as greyed out / hatched or similar in the full view when under PYMEVis or PYMEImage).
- Ideally work out and fix the particular display error we are running into here
- lots of more general pipeline debugging aids

zacsimile, 2020-11-10:
Here's the full traceback from after I click `Apply Text Changes`. I'm not convinced this error is coming from `labview_spooling_hacks`. I'd say it's an error adding items back into a deleted pane, but the pipeline rebuild does work if you hit `Apply Text Changes` so long as there's no `OutputModule` involved. 

```
DEBUG:PYME.recipes.localisations:no events found
WARNING:PYME.LMVis.pipeline:Spoofing focus from metadata: this usually implies an error in the input data (missing events) and results might vary
ERROR:PYME.LMVis.pipeline:Error trying to fudge focus positions
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/experimental/labview_spooling_hacks.py", line 32, in spoof_focus_from_metadata
    assert mdh['StackSettings.StepSize'] > 0  # np.arange will fail with a step size of 0
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/pipeline.py", line 283, in _processEvents
    position, frames = labview_spooling_hacks.spoof_focus_from_metadata(mdh)
  File "/Users/zachcm/Code/python-microscopy/PYME/experimental/labview_spooling_hacks.py", line 46, in spoof_focus_from_metadata
    sys.exc_info()[2])
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/six.py", line 702, in reraise
    raise value.with_traceback(tb)
  File "/Users/zachcm/Code/python-microscopy/PYME/experimental/labview_spooling_hacks.py", line 32, in spoof_focus_from_metadata
    assert mdh['StackSettings.StepSize'] > 0  # np.arange will fail with a step size of 0
AssertionError: Error spoofing focus events -  frames: [0], positions None
WARNING:PYME.LMVis.dyeRatios:Unknown dichroic "BLP02-561R"
/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py:758: RuntimeWarning: invalid value encountered in greater
  self.Index *= (self.resultsSource[k] > range[0])*(self.resultsSource[k] < range[1])
/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py:758: RuntimeWarning: invalid value encountered in less
  self.Index *= (self.resultsSource[k] > range[0])*(self.resultsSource[k] < range[1])
lw update
hl update
DEBUG:matplotlib.axes._base:update_title_pos
lw update
hl update
DEBUG:matplotlib.axes._base:update_title_pos
lw update
hl update
DEBUG:matplotlib.axes._base:update_title_pos
lw update
hl update
DEBUG:matplotlib.axes._base:update_title_pos
lw update
hl update
DEBUG:matplotlib.axes._base:update_title_pos
RecipeView._layout
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at /Users/robind/projects/buildbots/macosx-vm6/dist-osx-py36/Phoenix/ext/wxWidgets/src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed

The above exception was the direct cause of the following exception:

SystemError: <class 'wx._core.WindowDestroyEvent'> returned a result with an error set
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at /Users/robind/projects/buildbots/macosx-vm6/dist-osx-py36/Phoenix/ext/wxWidgets/src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed

The above exception was the direct cause of the following exception:

SystemError: <class 'wx._core.WindowDestroyEvent'> returned a result with an error set
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at /Users/robind/projects/buildbots/macosx-vm6/dist-osx-py36/Phoenix/ext/wxWidgets/src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed

The above exception was the direct cause of the following exception:

SystemError: <class 'wx._core.WindowDestroyEvent'> returned a result with an error set
DEBUG:matplotlib.axes._base:update_title_pos
DEBUG:PYME.recipes.localisations:no events found
WARNING:PYME.LMVis.pipeline:Spoofing focus from metadata: this usually implies an error in the input data (missing events) and results might vary
ERROR:PYME.LMVis.pipeline:Error trying to fudge focus positions
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/experimental/labview_spooling_hacks.py", line 32, in spoof_focus_from_metadata
    assert mdh['StackSettings.StepSize'] > 0  # np.arange will fail with a step size of 0
AssertionError

During handling of the above exception, another exception occurred:

[... 6 lines cut ...]
    raise value.with_traceback(tb)
  File "/Users/zachcm/Code/python-microscopy/PYME/experimental/labview_spooling_hacks.py", line 32, in spoof_focus_from_metadata
    assert mdh['StackSettings.StepSize'] > 0  # np.arange will fail with a step size of 0
AssertionError: Error spoofing focus events -  frames: [0], positions None
WARNING:PYME.LMVis.dyeRatios:Unknown dichroic "BLP02-561R"
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/vertical_recipe_display.py", line 274, in OnPaint
    x0, y0 = self._input_position(nd)
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/vertical_recipe_display.py", line 220, in _input_position
    xp, yp = item.Position
RuntimeError: wrapped C/C++ object of type foldingPane has been deleted
lw update
hl update
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/vertical_recipe_display.py", line 274, in OnPaint
    x0, y0 = self._input_position(nd)
[... cut ...]
DEBUG:matplotlib.axes._base:update_title_pos
```

In addition to documentation, I think it would be nice if we could just hide all of the output modules in PYMEVis. It will feel more user friendly if you don't have to check the documentation before you add a module.

Why is hitting `Apply Text Changes` a lousy way to trigger a re-run? The other way I can think of doing it is to invalidate the recipe. Is there an advantage to re-running via invalidation over hitting `Apply Text Changes`? Is there another way to trigger a re-run?

David-Baddeley, 2020-11-10:
The argument for keeping the output modules available is that @barentine (and others) wants to be able to put together a workflow on test data in PYMEVis and then run it with minimal modifications on the cluster (rather than, e.g. having to re-load and edit in bakeshop etc ...). This seems like a reasonable use case that we should support, we just need to find a way of making it clear that they don't have any meaning in a non-batch context.

That's a whole heap of errors (many super confusing), and I'm now not convinced that the output modules are at fault.

Re-running is triggered automatically by any change to the pipeline, so there should never be any need to manually "trigger" a re-run - if something isn't working, fixing the problem (e.g. a parameter issue) will automatically trigger a re-run. Conversely, re-running without fixing the error will never get a different result. This is low priority, but at some-point in the future, "apply text changes" would ideally get smarter and only apply changes based on a diff of the recipe.

David-Baddeley, 2020-11-10:
Super weird - the display issue is py3/wx4 specific

zacsimile, 2020-11-10:
Maybe we could put a gray box around `OutputModule`s in PYMEVis and if you mouseover it says "won't run here, will only run in batch mode"?

Most of these errors are just everyday PYMEVis errors (I see them no matter what I do). It might not be the fault of the `OutputModules`. However, this doesn't happen if I just add DBSCAN or another FilterTable, for example. For some reason I've only received this error with `OutputModules` HDF/CSV/etc. I haven't exhaustively tested all recipe tiles, though, so others could cause this as well.

David-Baddeley, 2020-11-10:
and clicking "apply text changes" again fixes it (alternates between being displayed and not)

David-Baddeley, 2020-11-10:
Important error is likely:

SystemError: <class 'wx._core.WindowDestroyEvent'> returned a result with an error set
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/david/python-microscopy/PYME/recipes/vertical_recipe_display.py", line 65, in _layout
    self.fp.DestroyChildren()
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at /opt/concourse/worker/volumes/live/b30868b5-2a7c-479c-4e29-27874bbc69a6/volume/wxpython_1547931175613/work/ext/wxWidgets/src/common/wincmn.cpp(478) in ~wxWindowBase(): any pushed event handlers must have been removed

and will likely be the result of one of the custom editors in the output modules not being torn down properly. It might be possible to put this in a `try: except:` block, although it's probably better to fix the editor.

David-Baddeley, 2020-11-10:
might have been fixed in wx recently:

see 
https://forums.wxwidgets.org/viewtopic.php?t=32138
http://trac.wxwidgets.org/ticket/14145

although it might be a related problem - points to potentially manually calling `UnInit()` as a workaround

David-Baddeley, 2020-11-10:
No idea why it's just the output modules though. Would be good to try something else which has an `Enum` trait to see if it's the traitsui Enum editor.

zacsimile, 2020-11-11:
Tried with some other recipes containing `Enum` traits and they work. It looks like the error gets thrown when it tries to rebuild `FitResults`. Like you said, this flips between working and not working depending on how many times you press `Apply Text Changes`.

zacsimile, 2020-11-11:
OK, `self.fp.DestroyChildren()` is what's causing the error. I don't see why, though. I don't think it's related to the editors. `CSVOutput`, for example, creates this error even though it has no custom editors. This leaves us with the option that `OutputModule` is the problem. The only difference between `OutputModule` and `ModuleBase`, which works, is that `OutputModule` insists `self.outputs` has a length of 0. I can't see this causing problems anywhere in the `vertical_recipe_display` code, though.

zacsimile, 2020-11-11:
But wrapping it in a `try` block definitely solves this.

```
            try:
                self.fp.DestroyChildren()
            except:
                pass
```

David-Baddeley, 2020-11-11:
Putting it in a `try` block is a somewhat unsatisfactory solution, as I suspect it's just masking an error which might come back to bite us later.  It would be nice if we could find out what is causing it. That said, I'm stumped and can't see anything which would make the output modules special. 

It's not fixed on wx=4.1

zacsimile, 2020-11-11:
I agree it's unsatisfactory, but it does work as a stop-gap. Seeing as I doubt many users will attempt this, we could leave it for now and fix it later. I leave it up to you.

Darn, I was going off the change log and thought that C++ reference counting stuff was fixed. Is that still broken, or do we think this issue is separate from that issue?

David-Baddeley, 2020-11-11:
Suspect it's separate - the reference counting would be likely to give us segfaults etc ... This looks like it's because someone has failed to remove an event handler.

zacsimile, 2020-11-17:
I've flipped my perspective on this a bit. I think we should just wrap this in a try block and call it good. I'm not sure this will bite us later. What do you see as the potential problem with us ignoring extra event handlers?

zacsimile, 2020-11-20:
Temporarily fixed by #604.

closed 2020-11-20 by zacsimile