python-microscopy/python-microscopy #607: [BUG]

Hannahmar, 2020-11-14:
I clicked on PYMEVis on my Windows 10 laptop, I went to File>Open raw/prebleached data, I opened a two channel TIFF, I went to GLComp and it was dark, Chan 0 was also blank. Images from both channels were mixed and appeared in Slices. Also what is GLComp?
I ran the Gaussian filter, and after a few minutes a blank window appeared and I couldn't see the image.
Not sure what I did but after zooming in and out, also adjusting display, I could see the image clearly on my second try.

[image]

David-Baddeley, 2020-11-15:
I suspect there are a number of things going on here

- File>Open raw/prebleached is designed for opening raw data that was used to generate a set of localisations that you have already opened. The "Chan 0" and "GLComp / composite" panes will the same zoom and position from the points viewer. This means that you might not get something sensible if: a) there are no points loaded or b) the metadata for either points or image is wrong (i.e. the software gets the relative scaling and/or position wrong). The voxelsize of the loaded data would be the most likely culprit here. The "Slices" tab is not effected as it ignores the scaling in the points view.
 - Most software saves multi-channel tiff stacks as a single XYZ stack with the colour channels interleaved (i.e. for a 3 colour tiff images 1,2,3 are the different channels of the 1st z slice, images 4,5,6 are the second z slice etc .... Unfortunately there is no standard way of indicating that this is what has been done, so you need to rely on something in the image metadata to tell you. How this is specified differs depending on what program saved the tifs. We currently tend to get the colour channels right if you save as OME TIFF (e.g. by using the bioformats exporter, rather than the standard save as tiff function in FIJI) or save as tiff from PYME, but don't do so well at guessing what is going on if the data doesn't have OME metadata. **Where did your multi-channel tiff come from?** 

For reference, the tabs in the image viewer are as follows:

**Chan 0, Chan 1 etc ... :** The individual colour channels, with zoom and position synced with the points viewer 

**Composite:** A composite of all colour channels, synced to the points viewer (only shown if the software detects multiple channels)

**Slices:** An unsynchronised view of the data (also supports x-z slices and a few extra annotation functions, which is why it's shown here)

**GLComp:** An experimental fast OpenGL based image viewer, synched to the points view. Set to replace both the individual channel views and the composite view once all the bugs have been ironed out .

Hannahmar, 2020-11-16:
Thank you for the reply! The tiff files were saved using the standard save as tiff function in FIJI.

zacsimile, 2021-05-18:
I think this might be fixed now. Can you confirm, @Hannahmar?

Hannahmar, 2021-05-19:
Not sure if problem is resolved yet. I opened a two channel TIF with open raw/prebleached data, nothing appeared in Chan0, I could see the images in Slices window. The file size is pretty big, 2.2 GB

[image]

I also tried using Gaussian filter and it was very slow, eventually I got an error message

 Error whilst running Processing>&Gaussian Filter
==============================================
python-microscopy=21.03.10
python=3.6.12, platform=win32
numpy=1.14.6, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "C:\Users\Hannah Tekleab\miniconda3\envs\pyme\lib\site-packages\PYME\ui\progress.py", line 121, in func
    return fcn(*args, **kwargs)
  File "C:\Users\Hannah Tekleab\miniconda3\envs\pyme\lib\site-packages\PYME\DSView\modules\filtering.py", line 59, in OnGaussianFilter
    filt_ims = [np.atleast_3d(gaussian_filter(self.image.data[:,:,:,chanNum].squeeze(), sigmas)) for chanNum in range(self.image.data.shape[3])]
  File "C:\Users\Hannah Tekleab\miniconda3\envs\pyme\lib\site-packages\PYME\DSView\modules\filtering.py", line 59, in <listcomp>
    filt_ims = [np.atleast_3d(gaussian_filter(self.image.data[:,:,:,chanNum].squeeze(), sigmas)) for chanNum in range(self.image.data.shape[3])]
  File "C:\Users\Hannah Tekleab\miniconda3\envs\pyme\lib\site-packages\PYME\IO\DataSources\BaseDataSource.py", line 138, in __getitem__
    r = np.concatenate([np.atleast_2d(self.getSlice(i)[keys[0], keys[1]])[:,:,None] for i in range(*keys[2].indices(self.getNumSlices()))], 2)
MemoryError

[image]

zacsimile, 2021-05-20:
This opens as a 2-channel image for me if I launch `dh5view` directly.

[image]

But not in raw/prebleached. I suspect it's related to this warning

```
/Users/zachcm/Code/python-microscopy/PYME/LMVis/imageView2.py:130: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
/Users/zachcm/Code/python-microscopy/PYME/LMVis/imageView2.py:318: PYMEDeprecationWarning: This will either disappear or change function as we move to a 5D data model. Use the explicit .data_xytc instead, or even better, change to using the 5D model as image.data_xyztc or image.voxels
  im = (self.image.data[x0_p:x1_p:step, y0_p:y1_p:step, self.do.zp, self.chan].squeeze().astype('f').T)
/Users/zachcm/Code/python-microscopy/PYME/LMVis/imageView2.py:322: wxPyDeprecationWarning: Call to deprecated item ImageFromData. Use :class:`Image` instead.
  imw = wx.ImageFromData(im.shape[1], im.shape[0], im.ravel())
```

zacsimile, 2021-05-20:
Similarly, Gaussian Filtering ran fine (albeit it took a few minutes, but it's a 2.2 GB image) in `dh5view`.

@Hannahmar, I recommend opening this image in `pymeimage` (same thing as `dh5view`) for now. Everything seems to work there. We'll get around to fixing it in raw/prebleached data shortly, but `pymeimage` will work for you right now.