python-microscopy/python-microscopy #608: [BUG]PYME development installs (python = 3.6) ImportError: DLL load failed: The specified module could not be found.

HelenSun09, 2020-11-16:
The installation does not seem to have worked. I got the following message when running "PYMEImage -t" or "visgui":
**ImportError: DLL load failed: The specified module could not be found.**followed the installation procedure as follows and got this message: 

I followed the instructions to install PYME development installs:
1) Install miniconda (version 3.8) 
2) conda config --add channels david_baddeley
conda create -n pyme pyme-depends python=3.6
conda activate pyme
4) Install visual build C++tools 2019 (16.8.1) 
5) git clone https://github.com/python-microscopy/python-microscopy.git
cd python-microscopy
python setup.py develop

I got the following message after installing PYME: 
Installed c:\users\helen sun\python-microscopy
Processing dependencies for python-microscopy==20.11.7
Searching for numpy==1.19.2
Best match: numpy 1.19.2
Adding numpy 1.19.2 to easy-install.pth file
Installing f2py-script.py script to C:\Users\Helen Sun\miniconda3\envs\pyme\Scripts
Installing f2py.exe script to C:\Users\Helen Sun\miniconda3\envs\pyme\Scripts

David-Baddeley, 2020-11-16:
This looks like it might be a duplicate of issue #551

Could you try the work-around mentioned there (`conda install blosc=1.19.0`) and see if that helps? Will try and get that into pyme-depends and rebuild by the end of the week.

HelenSun09, 2020-11-16:
"conda install blosc=1.19.0" fixed the issue.

closed 2020-11-16 by HelenSun09