python-microscopy/python-microscopy #609: [ENHANCEMENT] Rule 'nice' / priority

barentine, 2020-11-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I would be frustrated if I was running multiwell control flow using protocols/chained rules and localization/post-localization analysis from one well held up automatic ROI detection on a tile from another (and therefore the scope potentially sitting around waiting)

**Describe the solution you'd like**
- a 'nice' parameter for `IntegerIDRule` init, and the ruleserver doing a seachsort insert when adding a rule

**Are you prepared to work on it?**
of course, just checking in first

David-Baddeley, 2020-11-21:
I'm going to need some convincing - significantly increases complexity and my gut feeling is that if things are bogged down enough that this is needed then we have other more pressing issues to solve.

barentine, 2020-11-23:
Closing as David is right this does not look trivial to implement. I don't think this will ever be a particularly large problem for anyone who isn't pushing for speed/throughput ego-claims, and if they are the simpler solution would be to write faster code elsewhere / add to or upgrade the cluster

closed 2020-11-23 by barentine