python-microscopy/python-microscopy #613: [BUG] Measure objects has prominent placement but it's unclear how to use it

zacsimile, 2020-11-18:
**Describe the bug**
`Analysis>Measure objects` is one of the first things you see when you open the analysis menu, and one of the most likely things you'd want to use. The same tool(s) in ImageJ (`Analyze>Measure` or `Analyze>Analyze Particles`) measure area/perimeter of binary clusters. Presumably our Measure objects should also work on clusters. However, after clustering, this is the traceback

```
 Error whilst running Analysis>Measure objects
==============================================
python-microscopy=20.11.07
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/Extras/objectMeasurements.py", line 88, in OnMeasure
    ids = set(pipeline.mapping['objectID'].astype('i'))
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 923, in __getitem__
    return self.resultsSource[keys]
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 923, in __getitem__
    return self.resultsSource[keys]
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 723, in __getitem__
    return self.resultsSource[key][self.Index][sl]
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 879, in __getitem__
    res = np.array(self.resultsSource[key])[self.Index]
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 923, in __getitem__
    return self.resultsSource[keys]
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 923, in __getitem__
    return self.resultsSource[keys]
  File "/Users/zachcm/Code/python-microscopy/PYME/IO/tabular.py", line 468, in __getitem__
    raise KeyError('Key (%s) not found' % key)
KeyError: 'Key (objectID) not found'
``

It looks this means we need to create an image and then get segmented IDs from the image and use these as the clusters. We have no posted documentation on how to do this. If you just run `Analysis>Get segmented IDs from image`, you get the traceback

```
 Error whilst running Analysis>Get segmented IDs from image
==============================================
python-microscopy=20.11.07
python=3.6.10, platform=darwin
numpy=1.16.6, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/Extras/objectMeasurements.py", line 64, in OnGetIDs
    wx.CHOICEDLG_STYLE
TypeError: SingleChoiceDialog(): argument 4 has unexpected type 'generator'
```

We should fix and document (I think this has been done, even though it is not online) this pipeline. Is it possible to generalize this and make Measure objects work on all types of clusters (generated from DBSCAN or Corrections>Chaining)?

David-Baddeley, 2020-11-21:
Should definitely be fixed, generalised, and documented. It's one of the many breakages that occurred when we moved to shaders and layers.

zacsimile, 2020-11-21:
Can we, as a temporary fix, have it append `MeasureClusters3D` to the pipeline if it detects DBSCAN clustering? Might make part of the pipeline a bit more intuitive.

David-Baddeley, 2020-11-22:
This can't be fixed properly until we get to #524. Ultimately `MeasureClusters3D` should go away and be replaced with a 3D aware version of the code that gets called from `OnMeasure()`, and `OnMeasure would be altered to display a dialog which let you set the clustering variable. 

I'm not really prepared to replace the current functionality with that in `MeasureClusters3D` - best short-term fix would probably be to try a number of possible `objectID` keys, but tempted to have a look at moving at least part way to a proper fix.

zacsimile, 2020-11-22:
Trying different `objectID` key would be an excellent short-term fix. If you have the time, a proper fix would be great.

zacsimile, 2020-11-22:
I gave it a go in #633. The code appears to be quite old, and we're still injecting directly into the pipeline. Also, no plots come up and there's no coloring options, so it's not super clear how to use the object measurements. I'm now also leaning toward a proper fix.

I modified the triangulation call, but I'm not sure it's right. It appears to work on testing. The call used to be `scipy.spatial.Delaunay(object.ravel(),2)`, then it was `matplotlib.tri.Triangulation(object.ravel(),2)`. I changed it to `scipy.spatial.Delaunay(object)` since that seems to play nicest with line 133 (`obj = np.vstack([x[ind],y[ind]]).T`) and `gen3DTriangs.gen2DTriangsTF`. My uncertainty stems from not understanding what the `,2` did. What was this supposed to do?

David-Baddeley, 2020-11-22:
The code is very old ... - I think it was originally written using the `Delny` module before matplotlib and scipy had triangulation. When Delny became increasingly hard to install / unmaintained I switched everything to using matplotlib.tri.Triangulation (I think matplotlib had triangulation before scipy and/or it was a better match to the Delny API). Ultimately we'd need to go back and look at the matplotlib docs around the time that that code was altered, but it might be easiest to just figure out what works now. `object.ravel()` is a 1d array (i.e. passed to c code as a pointer), so I assume that the 2 is telling the underlying c code (probably qHull) to interpret the data as a set of 2d coordinates.

David-Baddeley, 2020-11-22:
Note - it's old enough that it doesn't really even inject into the pipeline per se, but rather into a separate objectMeasures dictionary which happens to be attached to the pipeline object.

David-Baddeley, 2020-11-22:
part 2 of a short term fix would be opening the measurements in the table viewer after #624

David-Baddeley, 2020-11-22:
Just for context for all of this, I've added a document on my cluster-analysis philosophy to messy_pyme_docs.

https://github.com/python-microscopy/messy_pyme_docs/blob/master/David%E2%80%99s%20cluster%20philosophy.docx

This should probably propagate to the full documentation at some point, but might be too conversational/informal for the paper.

zacsimile, 2020-12-21:
This is now usable on most (if not all) clustering and we have documentation in the PYMEVis user guide.

closed 2020-12-21 by zacsimile