python-microscopy/python-microscopy #614: [BUG] 3D jittered triangulation multiprocessing issue?

barentine, 2020-11-18:
**Describe the bug**

<details> <summary> traceback </summary>

```
ERROR:traits:Exception occurred in traits notification handler for object: <PYME.recipes.localisations.DensityMapping object at 0x7fb402ff5ca8>, trait: zBounds_items, old value: <undefined>, new value: TraitListEvent(index=1, removed=[59173.15015015015], added=[59173.15015015015])
Traceback (most recent call last):
  File "/opt/miniconda3/lib/python3.6/site-packages/traits/trait_notifiers.py", line 522, in _dispatch_change_event
    self.dispatch(handler, *args)
  File "/opt/miniconda3/lib/python3.6/site-packages/traits/trait_notifiers.py", line 616, in dispatch
    handler(*args)
  File "/opt/miniconda3/lib/python3.6/site-packages/traitsui/wx/list_editor.py", line 275, in update_editor_item
    for control in self.control.GetChildren():
RuntimeError: wrapped C/C++ object of type ScrolledPanel has been deleted
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
[... 1209 lines cut ...]
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().
Break on __THE_PROCESS_HAS_FORKED_AND_YOU_CANNOT_USE_THIS_COREFOUNDATION_FUNCTIONALITY___YOU_MUST_EXEC__() to debug.
```

</details>

**To Reproduce**
Steps to reproduce the behavior:
1. load something
2. run this [recipe](https://forum.image.sc/t/voronoi-tessellation-based-cluster-analysis/45548/2)
3. jittered triangulation result all zeros
4. check terminal and see traceback

David-Baddeley, 2020-11-18:
What platform - looks like OSX but have a suspicion that this will depend on which OSX version you are running. 

Ultimately I suspect this will come down to a pylab or wx import somewhere.

David-Baddeley, 2020-11-18:
Also: is this VisGUI or bakeshop? Might be worth checking if 'spawn workers for each core' is checked.

barentine, 2020-11-18:
PYMEVis, osx 10.15.7, wx 4.0.4

David-Baddeley, 2020-11-18:
Thanks, works for me under the (admittedly old) OSX 10.11.6

zacsimile, 2020-11-18:
Weird. I have the same macOS. I can run Jittered Triangulation just fine on an image. If I apply the recipe you reference (copy+paste the recipe text and hit Apply Text Changes) I get an entirely different error and PYMEVis crashes.

```
ERROR:traits:Exception occurred in traits notification handler for object: <PYME.recipes.localisations.DensityMapping object at 0x7f84d7731bf8>, trait: zBounds_items, old value: <undefined>, new value: TraitListEvent(index=0, removed=[42904.62962962963], added=[42904.62962962963])
Traceback (most recent call last):
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 522, in _dispatch_change_event
    self.dispatch(handler, *args)
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 616, in dispatch
    handler(*args)
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traitsui/wx/list_editor.py", line 269, in update_editor_item
    if (len(event.removed) != 1) or (len(event.added) != 1):
RecursionError: maximum recursion depth exceeded in comparison
Created colour filter with chans: ['chan0', 'chan1']
Created colour filter with chans: ['chan0', 'chan1']
Fatal Python error: Cannot recover from stack overflow.

Current thread 0x0000000107e41dc0 (most recent call first):
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/has_traits.py", line 2988 in trait_names
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/has_traits.py", line 1310 in trait_get
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 215 in outputs
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 104 in invalidate_parent
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 484 in dispatch
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 522 in _dispatch_change_event
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 550 in _notify_method_listener
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 476 in __call__
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 618 in notifier
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 234 in notify
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 349 in __setitem__
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 711 in __setitem__
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/localisations.py", line 86 in execute
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 641 in execute
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 488 in invalidate_data
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 106 in invalidate_parent
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 484 in dispatch
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 522 in _dispatch_change_event
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 550 in _notify_method_listener
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 476 in __call__
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 618 in notifier
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 234 in notify
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 349 in __setitem__
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 711 in __setitem__
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/localisations.py", line 86 in execute
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 641 in execute
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 488 in invalidate_data
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 106 in invalidate_parent
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 484 in dispatch
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 522 in _dispatch_change_event
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 550 in _notify_method_listener
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_notifiers.py", line 476 in __call__
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 618 in notifier
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 234 in notify
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 349 in __setitem__
[... cut ...]
  File "/Users/zachcm/miniconda3/envs/pyme/lib/python3.6/site-packages/traits/trait_list_object.py", line 711 in __setitem__
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/localisations.py", line 86 in execute
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 641 in execute
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 488 in invalidate_data
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 106 in invalidate_parent
  ...
Abort trap: 6

```

barentine, 2020-11-18:
thanks for having a go @zacsimile. I also hit that but hacked around by setting `pipeline.recipe.execute_on_invalidation=False` then manually `pipeline.recipe.execute()`. Obviously not a gold standard of 'this works', but if you do that I wonder if you hit the same fork exec error I do?

zacsimile, 2020-11-18:
Hi @barentine. With that hack, I get the same traceback as you, plus this extra little bit at the end.

```
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 641, in execute
    m.execute(self.namespace)
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 1176, in execute
    namespace[self.outputName] = self.filter(namespace[self.inputName])
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 1165, in filter
    filt_ims = [np.atleast_3d(self.applyFilter(image.data[:,:,:,chanNum].squeeze().astype('f'), chanNum, 0, image)) for chanNum in range(image.data.shape[3])]
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 1165, in <listcomp>
    filt_ims = [np.atleast_3d(self.applyFilter(image.data[:,:,:,chanNum].squeeze().astype('f'), chanNum, 0, image)) for chanNum in range(image.data.shape[3])]
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/processing.py", line 74, in applyFilter
    threshold = thresholding.otsu(data, nbins=self.n_histogram_bins, bin_spacing=self.bin_spacing)
  File "/Users/zachcm/Code/python-microscopy/PYME/Analysis/thresholding.py", line 151, in otsu
    "to have just one color {0}.".format(image.min()))
ValueError: threshold_otsu is expected to work with images having more than one color. The input image seems to have just one color 0.0.
```

David-Baddeley, 2020-11-21:
The otsu error message is super messed up too - maybe we should change it to be more infromative.

David-Baddeley, 2020-11-21:
A good candidate is the `matplotlib.tri` import in `CurrentRenderer._get_neighbour_dists()` in renderers.py or, maybe more likely, the similar imports in `visHelpers.rendJitTri` (these might implicitly try and use wx somehow). Working around this in an even moderately clean way is likely to be super hard (we need this import, and matplotlib will inherit the backend we set before `fork()`ing, so it might not be as simple as just setting the matplotlib backend before performing the import). 

Best bet would probably be to set the matplotlib backend to an offscreen one before we create our processes in `rendJitTriang()`. Another quick thing to try would be to make the import in question global in visHelpers (not really ideal in terms of keeping imports where they are needed, but probably useful for debugging).

David-Baddeley, 2020-11-30:
Sounds like it's this ...

https://bugs.python.org/issue33725

3/4 options:
- work out which library used core frameworks / objc libraries and nuke (what I tried above)
- tell multiprocessing to spawn instead e.g: 
    ```python
    if platform.system() == "Darwin":
        multiprocessing.set_start_method('spawn')
    ```
    and live with the performance hit.

- Set the `OBJC_DISABLE_INITIALIZE_FORK_SAFETY=YES` environment variable
- Test the OSX version and run single-process if we're on a dodgy OSX

David-Baddeley, 2022-12-14:
Tried this recently and couldn't replicate 

(py3.8, mac 12.1)

maybe someone else has fixed this in the meantime ...

David-Baddeley, 2022-12-14:
Will close, can reopen if we run into this again ...

closed 2022-12-14 by David-Baddeley