python-microscopy/python-microscopy #615: [ENHANCEMENT] andor sdk3 clara class

barentine, 2020-11-18:
**Is your feature request related to a problem? Please describe.**
labmate is going to run an older clara cooled ccd on a tc scope - looks like everything is there, though I'm assuming the move would be to add a clara file/class. I think it would be a good idea if we add somewhere in the PYMEAcquire doc a convenient list of what hardware (SDK's) we support already, similar to what @David-Baddeley included at python-data-acquisition meta.

**Are you prepared to work on it?**
yep

barentine, 2020-11-18:
note from discussion: copy Zyla and work from there

zacsimile, 2021-05-18:
I know @Hannahmar is also starting to code a camera. Now might be a good time to have a discussion about how we want to change the camera base class @David-Baddeley.