python-microscopy/python-microscopy #616: [BUG] Clip to Selection fails for small ROIs

lukasfue, 2020-11-18:
**Describe the bug**
I tried to 'Clip to Selection' on a small ROI, ~100x100 nm. The issue only seems to occur with ROIs that are smaller than ~400x400 nm It seems to partially work because the number of events updates and seems to be a reasonable number but the displayed points outside of the ROI don't disappear and I can no longer zoom in/out or pan around. 

**To Reproduce**
Steps to reproduce the behavior:
1. Load data
2. Attempt to clip to small ROI
3. See error:
Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\LMVis\gl_render3D_shaders.py", line 212, in OnPaint
    self.OnDraw()
  File "c:\users\laf62\python-microscopy\PYME\LMVis\gl_render3D_shaders.py", line 398, in OnDraw
    l.render(self)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\layers\base.py", line 118, in render
    return self.engine.render(gl_canvas, self)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\layers\pointcloud.py", line 28, in render
    n_vertices = vertices.shape[0]
AttributeError: 'NoneType' object has no attribute 'shape'

**Expected behavior**
Clips to region and works as normal

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
OS: Win10
Python version: 3.8.3
numpy version: 1.19.1

**Additional context**
- This was on two-color ratiometric data
- I am able to clip to other ROIs that are bigger than about 400x400 nm on the same data without issue

zacsimile, 2020-11-19:
From brief testing, this appears to be related to cropping a region that has no points in it rather than being size-specific.

lukasfue, 2020-11-20:
When I was doing it, the ROIs I selected had multiple points in them

zacsimile, 2020-11-20:
Can you provide a file with a region and/or screenshots? I'm able to select really tiny ROIs as long as there's still a point in them.

[image]

[image]

lukasfue, 2020-11-20:
Ahh, okay I think I figured out the specific case in which this error occurs. Its only with two color data and there has to be at least one point from each color in the ROI for it to successfully clip. If I have only one color in the ROI, even if there are many points and the ROI is large, it will not successfully clip.

David-Baddeley, 2020-11-21:
@zacsimile - this changes the fix we'd discussed - popping up a dialog whenever you select a ROI which only contains 1 channel would be pretty obnoxious..

zacsimile, 2020-11-21:
@David-Baddeley, I agree. Should we change the way ROI selection works so you can crop in regions where at least 1 color channel is present?

lukasfue, 2020-11-23:
Is there a technical reason why it wouldn't be possible/easy to make it so you can crop regions with just 1 color in it? I can imagine users wanting to do exactly that in cases of two-color data on different organelles for example when its less likely to have a lot of overlap between colors

David-Baddeley, 2020-11-24:
can hopefully close now...

closed 2020-11-24 by David-Baddeley