python-microscopy/python-microscopy #617: [BUG]PYMEClusterOfOne not working

HelenSun09, 2020-11-18:
**Describe the bug**
Setting icon
INFO:PYME.cluster.cluster_of_one:Launching data server: root=C:\Users\Helen Sun\PYMEData
INFO:PYME.cluster.cluster_of_one:Launching rule server
'C:\Users\Helen' is not recognized as an internal or external command,
operable program or batch file.
'C:\Users\Helen' is not recognized as an internal or external command,
operable program or batch file.
INFO:PYME.cluster.cluster_of_one:Launching node server
'C:\Users\Helen' is not recognized as an internal or external command,
operable program or batch file.
INFO:PYME.cluster.cluster_of_one:Shutting down cluster
Shutting down cluster
Traceback (most recent call last):
  File "C:\Users\Helen Sun\miniconda3\envs\pyme\Scripts\PYMEClusterOfOne-script.py", line 33, in <module>
    sys.exit(load_entry_point('python-microscopy', 'console_scripts', 'PYMEClusterOfOne')())
  File "c:\users\helen sun\python-microscopy\PYME\cluster\cluster_of_one.py", line 179, in main
    cluster.launch(gui=options.ui, clusterUI=options.clusterui)
  File "c:\users\helen sun\python-microscopy\PYME\cluster\cluster_of_one.py", line 100, in launch
    self._launch_ruleserver_ui()
  File "c:\users\helen sun\python-microscopy\PYME\cluster\cluster_of_one.py", line 77, in _launch_ruleserver_ui
    webbrowser.open_new_tab(list(ruleservers.values())[0])
IndexError: list index out of range

**To Reproduce**
Steps to reproduce the behavior:
1. Use an windows 10 system, python = 3.6
2. Go to miniconda3
3. conda activate pyme
4. PYMEClusterOfOne

barentine, 2020-11-18:
Thanks for posting @HelenSun09. This can happen if the computer name is too long, see #548 and #546. In another terminal, what happens if you run the command `PYMERuleServer`? If anything is printed, would you mind posting it here in full?

barentine, 2020-11-18:
Oh, sorry, ignore that. There's a space in your home directory file name, and we'll need to handle that in the PYMEClusterOfOne launch script at the very least.

HelenSun09, 2020-11-19:
Thank you so much for the fix @barentine.

David-Baddeley, 2020-11-23:
Not tested with spaces in path, but likely to work. Can be re-opened if needed

closed 2020-11-23 by David-Baddeley

HelenSun09, 2020-12-01:
The space in the directory still seems to be a problem. Cannot doing single-molecule localization in dh5view.  The localization is not going forward. 

**Describe the bug** 
**In the PYMEClusterOfOne prompt window:** 

Setting icon
INFO:PYME.cluster.cluster_of_one:Launching data server: root=C:\Users\Helen Sun\PYMEData
INFO:PYME.cluster.cluster_of_one:Launching rule server
INFO:ruleserver:Starting ruleserver on 127.0.0.1:0
INFO:numexpr.utils:NumExpr defaulting to 6 threads.
INFO:root:========================================
PYMEDataServer, running on python 3.6.12 |Anaconda, Inc.| (default, Sep  9 2020, 00:29:25) [MSC v.1916 64 bit (AMD64)]

INFO:root:Serving from directory: C:\Users\Helen Sun\PYMEData
INFO:root:Serving HTTP on 127.0.0.1 port 55392 ...
INFO:PYME.cluster.cluster_of_one:Launching node server
dict_values(['http://127.0.0.1:55388/'])
DEBUG:nodeserver:Starting to poll for tasks
nodeserver:INFO:Starting nodeserver on 127.0.0.1:0
INFO:nodeserver:Starting nodeserver on 127.0.0.1:0
127.0.0.1 - - [30/Nov/2020 20:42:18] code 404, message No handler for /favicon.ico
DEBUG:nodeserver:Launching worker processors
'C:\Users\Helen' is not recognized as an internal or external command,
operable program or batch file.
'C:\Users\Helen' is not recognized as an internal or external command,
operable program or batch file.
'C:\Users\Helen' is not recognized as an internal or external command,
operable program or batch file.
'C:\Users\Helen' is not recognized as an internal or external command,
operable program or batch file.
'C:\Users\Helen' is not recognized as an internal or external command,
operable program or batch file.
'C:\Users\Helen' is not recognized as an internal or external command,
operable program or batch file.
DEBUG:root:pytables open call: C:\Users\Helen Sun\PYMEData\FPLAM_Raw_Data\20201118\analysis\GlnA_SDP16_2_A_000_000_1.h5r
DEBUG:root:pytables file open: C:\Users\Helen Sun\PYMEData\FPLAM_Raw_Data\20201118\analysis\GlnA_SDP16_2_A_000_000_1.h5r
DEBUG:ruleserver:release_rule_tasks(ruleID = 000000-759aa8aaa7c14911992bfaf7cd2fc96f, release_start=10, release_end=9999)
DEBUG:root:H5RFile - closing: C:\Users\Helen Sun\PYMEData\FPLAM_Raw_Data\20201118\analysis\GlnA_SDP16_2_A_000_000_1.h5r
DEBUG:root:H5RFile - closed: C:\Users\Helen Sun\PYMEData\FPLAM_Raw_Data\20201118\analysis\GlnA_SDP16_2_A_000_000_1.h5r

**In the dh5view prompt window:** 
GenPanels
Creating datasource panel
DEBUG:PYME.LMVis.visCore:No Layout method
RecipeView._layout
Input datasources:  []
c:\users\helen sun\python-microscopy\PYME\LMVis\filterPane.py:111: wxPyDeprecationWarning: Call to deprecated item. Use InsertItem instead.
  ind = self.lFiltKeys.InsertStringItem(ind+1, key)
c:\users\helen sun\python-microscopy\PYME\LMVis\filterPane.py:112: wxPyDeprecationWarning: Call to deprecated item. Use SetItem instead.
  self.lFiltKeys.SetStringItem(ind, 1, '%3.2f' % value[0])
c:\users\helen sun\python-microscopy\PYME\LMVis\filterPane.py:113: wxPyDeprecationWarning: Call to deprecated item. Use SetItem instead.
  self.lFiltKeys.SetStringItem(ind, 2, '%3.2f' % value[1])
c:\users\helen sun\python-microscopy\PYME\ui\histLimits.py:289: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
DEBUG:root:numTotalFrames: 10000, currentFrameNum: 10
DEBUG:root:we have unpublished frames - push them
DEBUG:root:Successfully posted tasks
DEBUG:root:all tasks pushed, ending loop.
DEBUG:PYME.DSView.modules.LMAnalysis:use_cluster: newStyle=True, tq=None
Traceback (most recent call last):
  File "c:\users\helen sun\python-microscopy\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "c:\users\helen sun\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 712, in refresh_analysis
    self._refresh_analysis_cluster()
  File "c:\users\helen sun\python-microscopy\PYME\DSView\modules\LMAnalysis.py", line 769, in _refresh_analysis_cluster
    newNumAnalysed = int(queueInfo[self.analysisController.pusher._ruleID]['tasksCompleted'])
KeyError: '000000-759aa8aaa7c14911992bfaf7cd2fc96f'
Ev Idle
('points', 't')
refreshed
DEBUG:PYME.DSView.modules.LMAnalysis:use_cluster: newStyle=True, tq=None

**To Reproduce**
Steps to reproduce the behavior:

1. Use an windows 10 system, python = 3.6 
2. Go to miniconda3 -> conda activate pyme -> PYMEClusterOfOne
3. Open another miniconda3 prompt -> conda activate pyme ->dh5view -> "LManalysis" module -> Threshold = 0.7; subtract background in fit; Lat GaussFitFR