python-microscopy/python-microscopy #621: [BUG] Can't open recipe tab in image views started from VisGUI

csoeller, 2020-11-19:
**Describe the bug**

When you have an image viewer open in VisGUI (either via `Generate` created (rendering) or by `File>Open raw/prebleach data`), trying to request a recipe viewer tab fails without opening bug window, with error going to STDERR as shown further below.

**To Reproduce**
Steps to reproduce the behavior:
1. Open image in VisGUI, e.g. `File>Open Raw/Prebleach Data`
2. Request Recipe tab with `Modules/recipes`
3. Nothing happens in window
4. See error generated into STDERR:

```
Traceback (most recent call last):

  File "/Users/XXX/Documents/src/PYME-src/python-microscopy/PYME/DSView/dsviewer.py", line 246, in OnToggleModule

    self.LoadModule(mn)

  File "/Users/XXX/Documents/src/PYME-src/python-microscopy/PYME/DSView/dsviewer.py", line 253, in LoadModule

    modules.loadModule(moduleName, self)

  File "/Users/XXX/Documents/src/PYME-src/python-microscopy/PYME/DSView/modules/__init__.py", line 118, in loadModule

    ret = mod.Plug(dsviewer)

  File "/Users/XXX/Documents/src/PYME-src/python-microscopy/PYME/DSView/modules/recipes.py", line 247, in Plug

    dsviewer.create_overlay_panel()

  File "/Users/XXX/Documents/src/PYME-src/python-microscopy/PYME/DSView/dsviewer.py", line 274, in create_overlay_panel

    self.overlaypanel = OverlayPanel(self, self.view, self.image.mdh)

AttributeError: 'DSViewFrame' object has no attribute 'view'
```

**Expected behavior**
Open Recipe tab without any complaints.

**Screenshots**

[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
python-microscopy=20.11.07
python=2.7.17, platform=darwin
numpy=1.16.6, wx=3.0.2.0 osx-cocoa (classic)

**Additional context**
Seems a regression that used to work. Bug reproduced with latest github code as of time of bug submission.

zacsimile, 2020-11-19:
Hi @csoeller. I'm not 100% sure how this is supposed to work, we'll have to ask @David-Baddeley, but you can work around this by clicking `Modules>arrayView` before `Modules>recipes`.

csoeller, 2020-11-19:
Hi @zacsimile, thanks for the work around, this helps. I would imagine a proper fix would be preferable long term but this is good enough for now!

David-Baddeley, 2020-11-20:
*Medium term fix:* comment out `dsviewer.create_overlay_panel()`  in `DSView.modules.recipes`(the actual use cases for requiring that are super rare and shouldn't impact anyone).

*Long term:* This should all go away once we integrate overlays with the openGL views and unify the various different viewers.

csoeller, 2020-11-20:
Good to know, @David-Baddeley. What is the suggestion for now? I can leave as is and live with the work around but leave issue open so we know it needs to be checked for correct working eventually?

zacsimile, 2020-11-20:
@csoeller, I've implemented David's medium-term fix in #623, which should prevent you from needing to do the workaround. Once merged, we can probably close this issue if we find a way of keeping track of all these details we'll need to address during viewer unification. Maybe there should be a separate "viewer unification" issue that we update as we find these things?

csoeller, 2020-11-20:
@zacsimile , thanks, sounds very good. Happy to go with this idea as long as the viewer unification issue is recorded elsewhere as otherwise we may well loose the record going forward.

David-Baddeley, 2020-11-21:
Closing following merge of #623. Unification stuff is mentioned in roadmap (which should be good enough).

closed 2020-11-21 by David-Baddeley