python-microscopy/python-microscopy #622: [BUG] Running pointwise colocalization on a 3-color image throws non-descriptive error

zacsimile, 2020-11-20:
**Describe the bug**
If we run `Analysis>Pointwise Colocalisation` on a 3-color image, we get the following traceback.

```
Traceback
=========
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/zachcm/Code/python-microscopy/PYME/LMVis/Extras/pointwiseColoc.py", line 25, in OnPointwiseColoc
    distColoc.calcDistCorr(self.visFr.pipeline.colourFilter, *(self.visFr.pipeline.colourFilter.getColourChans()[::1]))
  File "/Users/zachcm/Code/python-microscopy/PYME/Analysis/Colocalisation/distColoc.py", line 47, in calcDistCorr
    hAA = DistHist.distanceHistogram(xA, yA, xA, yA, nbins, binsize)
TypeError: an integer is required (got type str)
```

This is because we pass all 3 color channel names as `*args`, and so the variable `nbins` gets replaced by a string.

**To Reproduce**
Steps to reproduce the behavior:
1. Open any 3-channel (or greater) file
2. `Analysis>Pointwise Colocalisation`

**Expected behavior**
This is open to discussion. I can see us addressing this in a number of ways.

- Tell the user to use a 2-color image and don't run (seems unnecessarily restrictive).
- Clip the length of `*args` passed to 2. This will work, but the user will not be able to analyze the color channels of their choosing.
- Have a `PYME.DSView.modules.coloc.ColocSettingsDialog` pop up to allow the user to select the two color channels of their choice.
- Extend the capabilities of `Pointwise Colocalisation` to run through every combination of channels and pop up multiple graphs.

Curious to hear your thoughts on this, @David-Baddeley.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOs 10.15.7
 - Python version: 3.6.10
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2020-11-21:
Popping up a chooser is probably the best, if also the most work. At a pinch, truncating along with printing/displaying a warning might be acceptable I don't really like the do all possible combinations option.

zacsimile, 2020-11-21:
Choose implemented in #628.

closed 2020-11-23 by David-Baddeley