python-microscopy/python-microscopy #627: [BUG] Spaces cause lines to cut off early in modal dialogs in wx4

zacsimile, 2020-11-21:
**Describe the bug**
Spaces seem to create a problem for us in modal dialogs. We've run into this before (#432, #388). Now I've noticed it on `Show Shortcuts`. Rather than fix this for another specific instance, I suggest we figure out why the heck spaces cause lines to cut off early in wx 4.0.4. This is clearly a general problem and it makes things look messy.

[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15.7
 - Python version: 3.6.10
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: 4.04

zacsimile, 2021-03-16:
I can fix some of it by adding proportion and expand flags, but the left side text is still cut off. This is the closest I've got to perfection and I think I've now spent too much time on what I'm increasingly convinced is a wx bug.

```
def add_shortcut(self, table, short_cut, action):
        table.Add(wx.StaticText(self, label=short_cut), 1, wx.EXPAND, 5)
        table.Add(wx.StaticText(self, label=action), 1, wx.EXPAND, 5)
```

[image]

zacsimile, 2021-05-14:
This appears to be fixed in python 3.8/wxpython 4.1.

David-Baddeley, 2021-12-05:
I'm going to close this as it appears to be a wx issue, now fixed, and would be counterproductive to spend too much time patching around.

closed 2021-12-05 by David-Baddeley