python-microscopy/python-microscopy #631: [BUG] Datasources disappear if you mess up parameters in your recipe

zacsimile, 2020-11-22:
**Describe the bug**
If you screw up parameters in your recipe so that one of your outputs returns 0 points, the datasources after this will disappear. You won't know what the problem is until you switch to the `Pipeline Recipe` tab.

[image]

**Expected behavior**
- Datasource stay selectable in Data pipeline output, but return no points
- There's a red box around the module in the Data pipeline view as well as the Pipeline recipe view

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15.7
 - Python version: 3.6.10
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2020-11-22:
I'd argue that the datasources disappearing is the correct behaviour - they are not present if the pipeline doesn't execute. 

Adding a red box in the vertical view and /or greying out the data that is not present in a pipeline in both recipe views would be desirable.

David-Baddeley, 2020-11-22:
This falls into the general 'making recipes easier to debug' section of the roadmap.

closed 2020-12-21 by zacsimile