python-microscopy/python-microscopy #632: [BUG] glCanvas doesn't refresh after changing selectedDataSource

zacsimile, 2020-11-22:
**Describe the bug**
If you change the selected data source under Data pipeline output, you need to then change a layer property as well to get the canvas to refresh.

**To Reproduce**
Steps to reproduce the behavior:

[image]

[image]

[image]

**Expected behavior**
glCanvas should refresh when `selectedDataSource` changes

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15.7
 - Python version: 3.6.10
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: 4.0.4

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2020-11-23:
NOTE- only when the column / point colour you have selected is not in the new datasource.

zacsimile, 2020-12-20:
Maybe we can add a check when you change datasource that flips the colouring to a valid parameter (e.g. `z` or `t`) if the previous data source `vertexColour` isn't found.

zacsimile, 2021-03-16:
Hmm, I just tried to recreate this issue and the Colour selection automatically adjusted. Did you fix this @David-Baddeley?

David-Baddeley, 2021-03-16:
Not that I'm aware of

closed 2022-05-11 by zacsimile