python-microscopy/python-microscopy #643: [BUG] conda pyme 2.7 install with pyme-depends still pulls in traitsui 7

csoeller, 2020-11-24:
**Describe the bug**

This bug relates to the root cause of #567 which was closed. However, the root cause is still present in current conda installs as I discovered with a fresh install of latest pyme + python 2.7 on windows 10.

Took me a while to find the instructions again. Before closing this issue we should make sure it is really gone from current conda installs.

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest github pyme into a 2.7 conda environment, i.e.
2. `conda create --name pyme-default-plain python=2.7 pyme-depends`
3. Start visgui and try to do things
4. Falls over with traitsui issues

**Current Workaround**
Force downgrade of traitsui to 6.X, i.e.

`conda install traitsui=6`

**Proper fix**

??

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - Win10
 - Python version: 2.7.X
 - numpy version: ??
 - wxpython version [if a UI bug]: ??

David-Baddeley, 2020-11-24:
I haven't been able to get conda to successfully build packages for py27 for a while, so the py27 packages for pyme-depends and python-microscopy are super out of date. Best way of installing on 27 at present is probably to pip-install the dependencies instead (and/or manually downgrade things like traits).

David-Baddeley, 2020-11-24:
Note - officially dropping 27 support in Jan (although might try and retain code compatibility for source/manual installs for another few months until I'm sure all the microscope hardware stuff works on py3)

csoeller, 2020-11-24:
Ok, point taken, probably still good to leave this issue open until it is resolved or conda with py27 is officially deprecated.

I will give the pip route a spin in a fresh environment. Is it pip with requirements.txt?

Doing all this while working up a win10 machine to try camera acqusition on. I will try py3 on it as well but need to figure out a number of things for that. Jan could be tight for us...

David-Baddeley, 2020-11-24:
When you do make the move to py3 on acquisition I've got a bunch of py3 fixes to acquisition hardware sitting un-pushed on the instrument PC in Auckland, which might be helpful. 

Figure it would take a day or two to get py27 conda packages building again and it's hard to find/justify the time at the moment. If you want to have a go, knock yourself out - everything should be in the pyme-conda-recipes repo, and the workflow files in the .github folder are a good start for the steps which are needed.

You want to start with a fresh conda environment with conda and conda-build, and I'd try pyme-depends as the highest priority (just getting that building on 2.7 would fix this particular issue). Last time I tried, however, conda just spun it's wheels for ever trying to resolve the package set.

csoeller, 2020-11-25:
Thanks @David-Baddeley, getting access to the py3 fixes for acqusition would be nice. I am gradually testing this, projection is it will take me a few months to fully work through.

Re installation and conda + 2.7, I suppose I am also lacking round tuits and I am as yet unfimilar with conda build processes. So I may not attempt that too soon, rather than updating our installation instructions with appropriate workarounds.

I might carry the installation instructions thread to image.sc.

David-Baddeley, 2021-01-28:
Going to close this, saying we no longer officially support conda installs on py27.  Note the py3 acquisition fixes are in.

closed 2021-01-28 by David-Baddeley