python-microscopy/python-microscopy #644: [BUG] building current PYME on 2.7 with MS Visual C for Python 2.7 triggers error missing 'stdint.h'

csoeller, 2020-11-24:
**Describe the bug**
I tried building latest PYME in a fresh conda environment on windows 10 using python 2.7 and [MS Visual C for Python 2.7](https://www.microsoft.com/en-us/download/details.aspx?id=44266). This triggered a missing `stdint.h` error.

Error message:

```
C:\Users\XXX\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\amd64\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -IPYME\experimental -IC:\ProgramData\Anaconda2\envs\pyme-default-plain\lib\site-packages\numpy\core\include -Iwin_incl -IC:\ProgramData\Anaconda2\envs\pyme-default-plain\lib\site-packages\numpy\core\include -IC:\ProgramData\Anaconda2\envs\pyme-default-plain\include -IC:\ProgramData\Anaconda2\envs\pyme-default-plain\PC -IC:\ProgramData\Anaconda2\envs\pyme-default-plain\include -IC:\ProgramData\Anaconda2\envs\pyme-default-plain\PC /TcPYME\experimental\_triangle_mesh.c /Fobuild\temp.win-amd64-2.7\Release\PYME\experimental\_triangle_mesh.obj -O3 -fno-exceptions -ffast-math -march=native -mtune=native
cl : Command line warning D9002 : ignoring unknown option '-O3'
cl : Command line warning D9002 : ignoring unknown option '-fno-exceptions'
cl : Command line warning D9002 : ignoring unknown option '-ffast-math'
cl : Command line warning D9002 : ignoring unknown option '-march=native'
cl : Command line warning D9002 : ignoring unknown option '-mtune=native'
_triangle_mesh.c
c:\programdata\anaconda2\envs\pyme-default-plain\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
c:\python-support\python-microscopy-git\pyme\experimental\triangle_mesh_utils.h(4) : fatal error C1083: Cannot open include file: 'stdint.h': No such file or directory
```

**To Reproduce**
Steps to reproduce the behavior:
1. As above, try building PYME, e.g. with `python setup.py develop`
2. Hit the error

**Expected behavior**
Just complete the build without errors.

**Workaround adopted**

The following site has details that I followed and that solved the problem: https://stackoverflow.com/questions/44865576/python-scikit-image-install-failing-using-pip

Essentially, what the link does is letting you download a working `stdint.h` and put it where the compiler finds it.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10

David-Baddeley, 2020-11-24:
Hmm - we've solved that for another module (can't recall exactly how or which). Looping @zacsimile in as he wrote triangle_mesh_utils. 

Another potential workaround might be to use the community edition of the full MS VisualC instead.

zacsimile, 2020-11-24:
I compile this with Visual Studio Build Tools 2019 on Windows 10: https://visualstudio.microsoft.com/downloads/. I recommend giving that a go.

csoeller, 2020-11-25:
Thanks @zacsimile. For the record, I found a good windows compiler overview for python at https://wiki.python.org/moin/WindowsCompilers. This includes a more direct link to [Microsoft Build Tools for Visual Studio 2019](https://www.visualstudio.com/downloads/#build-tools-for-visual-studio-2019) which is presumably what you refer to above.

I will give that a try and report back.

I suppose I am also wondering if there are already updated build instructions in the docs for py3 (and possibly py27)? I appreciate this is a moving traget and aim to update our local instructions that we keep on github as I learn more.

zacsimile, 2020-11-25:
Thanks @csoeller! So, the Build tools for 2019 were the default built tools that miniconda (or was it anaconda?) told me to install during setup. A window just popped up in the middle of the installation process that said download and install this now. If that's changed, then we may need to add an install link to our docs.

csoeller, 2020-11-25:
Ok, thanks @zacsimile, good points. I guess I (and maybe others?) got tripped up as I was working with an existing conda/minconda installation that was from a time where such finesse was apparently not implemented. Based on that experience, it would always be good to have some fallback type instructions for those unlucky ones?

zacsimile, 2020-11-25:
It would be nice! It might be hard to make a general recommendation. I have no problem listing a recommendation for Build Tools 2019 for Windows 10, but if folks have a Windows 7 or a Mac or Linux machine with this same problems, the fixes can be quite different. Maybe this belongs in a separate Troubleshooting section rather than in the main install docs, and we can update the troubleshooting on a case-by-case basis? Alternatively we could move this conversation to image.sc and log it there as our troubleshooting.

csoeller, 2020-11-25:
I already did move it to image.sc and we can probably carry on there. I'll close this one now.

closed 2020-11-25 by csoeller