python-microscopy/python-microscopy #650: [ENHANCEMENT] supress matplotlib etc DEBUG info in dh5view (like VisGUI already does)

csoeller, 2020-11-30:
**Is your feature request related to a problem? Please describe.**

On both macOS and win10 py3 installs of PYME dh5view produces huge numbers of PYME unrelated debug statements, see screenshot:

[image]

**Describe the solution you'd like**
VisGUI has the following statements to suppress this:

```
logging.getLogger('matplotlib.font_manager').setLevel(logging.ERROR) #clobber unhelpful matplotlib debug messages
logging.getLogger('matplotlib.backends.backend_wx').setLevel(logging.ERROR)
logging.getLogger('PIL.PngImagePlugin').setLevel(logging.ERROR)
```

**Describe alternatives you've considered**
NA

**Are you prepared to work on it?**
Sure, some guidance where to insert the relevant statements would be great.

David-Baddeley, 2020-11-30:
Should be as simple as copying lines 57-61 of VisGUI.py to dsviewer.py (around line 48). The only caveat is to make sure that matplotlib and PIL have been imported before you set the log level on their loggers.

csoeller, 2020-11-30:
Cool, will raise a PR soonish...

David-Baddeley, 2020-11-30:
should be fixed with #651

closed 2020-11-30 by David-Baddeley