python-microscopy/python-microscopy #652: [BUG] clusterIO.get_file OSErrors more prevalent after get_advertise_services performance regression fix

barentine, 2020-11-30:
**Describe the bug**
Performance is generally rescued by #645s, however we are dropping frames in localization analysis (and recipe work) 
[image]. Can be particularly annoying for recipe PZF sources as one frame will fail out the whole thing.

```
Error in rule: 000003-ea8d53ae02ec418faf2da28b9d1d53c2 running task: 204 on elektra:12366
taskDescr:
----------
{'id': '000003-ea8d53ae02ec418faf2da28b9d1d53c2~204', 'type': 'localization', 'taskdef': {'frameIndex': '204', 'metadata': 'PYME-CLUSTER://villains/Bergamot/2020_11_25/EMC06.05.S1/000/analysis/25_11_series_G.h5r.json'}, 'inputs': {'frames': 'PYME-CLUSTER://villains/Bergamot/2020_11_25/EMC06.05.S1/000/25_11_series_G.pcs'}, 'outputs': {'fitResults': 'http://10.150.7.3:15348/__aggregate_h5r/Bergamot/2020_11_25/EMC06.05.S1/000/analysis/25_11_series_G.h5r/FitResults', 'driftResults': 'http://10.150.7.3:15348/__aggregate_h5r/Bergamot/2020_11_25/EMC06.05.S1/000/analysis/25_11_series_G.h5r/DriftResults'}}
Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 308, in computeLoop
    res = task()
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 417, in __call__
    bufferManager.updateBuffers(md, self.dataSourceModule, self.bufferLen)
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 113, in updateBuffers
    md['Camera.ElectronsPerCount'])
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 41, in __init__
    self.slice_shape = [int(d) for d in self.data_buffer.dataSource.getSliceShape()]
  File "/home/ubuntu/code/python-microscopy/PYME/IO/DataSources/ClusterPZFDataSource.py", line 78, in getSliceShape
    self.fshape = self.getSlice(0).shape
  File "/home/ubuntu/code/python-microscopy/PYME/IO/DataSources/ClusterPZFDataSource.py", line 71, in getSlice
    sl = PZFFormat.loads(clusterIO.get_file(frameName, self.clusterfilter))[0]
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 806, in get_file
    raise IOError("Specified file could not be found: %s" % filename)
OSError: Specified file could not be found: Bergamot/2020_11_25/EMC06.05.S1/000/25_11_series_G.pcs/frame00000.pzf
```

**To Reproduce**
Steps to reproduce the behavior:
1. use a non-local cluster for any kind of ClusterPZFDataSource analysis

David-Baddeley, 2020-11-30:
I'd run into this locally on PYMEClusterOfOne and have an uncommitted candidate debugging fix / mitigation sitting on my laptop. As far as I can tell problem is as follows:

- Data server can get bogged down if there is lots of traffic
- this results in a long round trip time
- `is_port_open` times out and dataserver gets marked as absent

The candidate fix increases the timeout on `is_port_open()` from 1 s to 10 s, which seems to fix things in practice, but is potentially not robust. Will commit and push, but we might need to look at other solutions as well.

David-Baddeley, 2020-11-30:
Other possible mitigations:

- temporarily remove `is_port_open()` call from zeroconf nameserver code. This makes the code less robust, especially on an uncontrolled network where people might be caching and proxying mDNS requests, but it was previously arguably good enough. The situations where this checking is really valuable for zeroconf name resolution are mostly `ClusterOfOne` type use cases where we now use `sqlite_ns` instead.
- add something like a 3-strikes rule (i.e. `is_port_open` has to fail 3 times in succession before it gets removed). Goal would be to make the probability of erroneously marking a service as missing vanishingly low.

David-Baddeley, 2020-11-30:
I've also added 3-strikes on the zeroconf unregistration and made sqlite checking only happen on startup, not whenever you spawn a thread.

barentine, 2020-12-02:
Thanks @David-Baddeley . I am WFH today but will close and reopen if needed.

closed 2020-12-02 by barentine