python-microscopy/python-microscopy #653: [BUG] GLComp tab broken in Gaussian renderer on Win10

zacsimile, 2020-11-30:
**Describe the bug**
GLComp window displays as all white on Win 10 after generating Gaussian. Other tab is fine. Traceback below.

```
Traceback (most recent call last):
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\LMVis\shader_programs\shader_program.py", line 103, in use
    glUseProgram(self._program)
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\OpenGL\error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
        err = 1281,
        description = b'invalid value',
        baseOperation = glUseProgram,
        cArguments = (7,)
)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\OpenGL\latebind.py", line 41, in __call__
    return self._finalCall( *args, **named )
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\LMVis\gl_render3D_shaders.py", line 213, in OnPaint
    self.OnDraw()
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\LMVis\gl_render3D_shaders.py", line 400, in OnDraw
    l.render(self)
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\LMVis\layers\base.py", line 118, in render
    return self.engine.render(gl_canvas, self)
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\LMVis\layers\image_layer.py", line 82, in render
    with self.shader_program:
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\LMVis\shader_programs\DefaultShaderProgram.py", line 57, in __enter__
    self.get_shader_program().use()
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\LMVis\shader_programs\shader_program.py", line 105, in use
    exception = GLProgramException(glGetProgramInfoLog(self._program))
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\OpenGL\latebind.py", line 61, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\OpenGL\GL\VERSION\GL_2_0.py", line 332, in glGetProgramInfoLog
    length = int(glGetProgramiv(obj, GL_INFO_LOG_LENGTH))
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\OpenGL\latebind.py", line 45, in __call__
    return self._finalCall( *args, **named )
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\OpenGL\wrapper.py", line 666, in wrapperCall
    raise err
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\OpenGL\wrapper.py", line 659, in wrapperCall
    result = wrappedOperation( *cArguments )
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\OpenGL\platform\baseplatform.py", line 402, in __call__
    return self( *args, **named )
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\OpenGL\error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
        err = 1281,
        description = b'invalid value',
        baseOperation = glGetProgramiv,
        pyArgs = (
                7,
                GL_INFO_LOG_LENGTH,
                <object object at 0x0000028E74B64520>,
        ),
        cArgs = (
                7,
                GL_INFO_LOG_LENGTH,
                array([0], dtype=int32),
        ),
        cArguments = (
                7,
                GL_INFO_LOG_LENGTH,
                array([0], dtype=int32),
        )
)
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10
 - Python version: 3.6.12
 - numpy version: 1.14.6
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2020-11-30:
what PyOpenGL version?

David-Baddeley, 2020-11-30:
(is this the executable installer??)

zacsimile, 2020-11-30:
This is the executable installer

zacsimile, 2020-11-30:
`pyopengl==3.1.1a1`

David-Baddeley, 2020-11-30:
hmm, that's the same one I have on my machine - looks like it hasn't been updated in a while, but also rules that out as a possible cause.

zacsimile, 2020-12-07:
No longer broken. This was an error associated with the single-click installer, so maybe there was just a compile issue before?

zacsimile, 2020-12-07:
Just kidding, still broken. I only get this error when open `hdf` files, and it's a problem with all of the renderers. Strangely the GLComp window works fine if I render from a `mat` or `csv` file.

closed 2020-12-07 by zacsimile