python-microscopy/python-microscopy #655: [ENHANCEMENT] Allow mapping on import

lukasfue, 2020-11-30:
**Is your feature request related to a problem? Please describe.**
Yes, There are certain variables that I need to map, but currently I can only do so in the shell.

**Describe the solution you'd like**
Upon initial import of data from a table (such as a MATLAB matrix), a mapping window pops up and allows me to convert variables from, in my case, Vutara/4Pi names to names Visgui will recognize. One of these is Vutaracrlb which equals error_x * sqrt(2)/pixelsize. However, error_x is what I need to define and python can't do algebra on its own apparently, so I cannot map it in this initial mapping window. It would be nice to be able to map variables that have more complicated definitions upon initial import, like error_x in this case, followed by saving through the gui.

My current way around this is to map the variables I can (x, y, z, t, nPhotons), and then go into the shell and use the lines: 

from PYME.IO.tabular import MappingFilter
mf = MappingFilter(pipeline.selectedDataSource, error_x='vutaracrlb / sqrt(2) * 128')
mf.to_hdf('myfile.hdf')

barentine, 2020-11-30:
Hey @lukasfue , not sure what you mean by Python can't do algebra on it's own, but does adding a mapping recipe module in the pipeline recipe tab work for you?
[image]

[image]

barentine, 2020-11-30:
`File > Save filtered localizations` is the GUI route to save an hdf file, after you've added the mapping to the pipeline recipe and made sure the end of it is your selected datasource

lukasfue, 2020-11-30:
So I just tested this. Do I understand this correctly that it just made a new datasource with everything the same except now it also has the error_x variable correctly mapped?

David-Baddeley, 2020-11-30:
The `MappingFilter` is a pass-through datasource, so in someways it's a new data source, but it has effectively zero cost (if you haven't specified a mapping it just gets the column from the source data).

David-Baddeley, 2020-11-30:
I think we should boil this down to 3 questions:
- Is this common enough that we need a simple GUI way of doing this? 
- If so, what should that look like:
      - The ability to set mappings in the import dialog?
      - A mapping filter in the pipeline by default?
- Should we check for `"vutaracrlb"` and specifically add this mapping 

With the current code you can load and save custom recipes with VisGUI. You could add a suitable mapping filter in the recipe tab (I'd put it either before `Pipelineify` or between `Piplelineify` and `ProcessColour`), save that recipe (effectively a customised pipeline that you'd use with all your data of a given type), and then launch VisGUI with `PYMEVis -r my_custom_pipeline.yaml data.csv`. There's a chance that this might be good enough for the majority of your usage ...

For a bit of context for the above, the input filters and pipelineify both currently return `MappingFilter` objects, so it would be reasonably simple to add that mapping either with a special case when it sees the `"vutaracrlb"` key or something more generic in the import dialog (which would be harder to implement).

David-Baddeley, 2020-11-30:
Adding the ability to set mappings on import or adding a mapping filter to the pipeline by default would both be flexible solutions, but would add clutter that might not be needed for most users.

zacsimile, 2020-12-11:
The ability to create IO plugins might solve this.

csoeller, 2020-12-11:
I'd strongly support something akin to IO plugins, seemless import of a number of formats (which we can add to over time) would be great and a real usability boon.

David-Baddeley, 2020-12-11:
IO plugins are definitely a good idea, but I don't think the address this specific problem. The most logical way of implementing IO plugins is to use the current code to handle the extensions we already support, and then try plugins which are registered by file extension for unknown extensions. In this case the files are very standard .csv, which we already handle, and there is no simple and robust way of detecting that they should be be handled by a plugin instead. IE - we would still open them using our builtin csv loader even if we had plugins.

We could in principle have plugins populate a `File-->Import` menu and allow people to manually select the plugin they want to use, but this won't work on the command line or for batch processing. To me this feels like a pretty broken solution.

barentine, 2020-12-11:
Maybe not the right spot for this discussion (filed [an issue on pyme-plugin](https://github.com/python-microscopy/pyme-plugin/issues/6), but realized I wasn't actually 'watching' it to receive notifications until today, nor is anyone else), but @David-Baddeley I think I could take a quick crack at this. Simplest would be to have a `.PYME/config/io.yaml` mapping extensions to an import method.

I really like the idea of a `File --> Import` menu item which would then allow you to force a load route. One correct/canonical way to load a given file extension would be grand, but I think our crusade should be half-hearted there since interoperability with other packages/etc is at stake, and we very clearly know even 'standards' in our field change over time

barentine, 2020-12-11:
re: forcing a load route from command line or in batch processing - if we do it right you could config the priority of the loader relative to 0 being default PYME

csoeller, 2020-12-11:
Final comment from me in this thread, as agree it is off topic. I am with @barentine that a pragmatic solution with manual import selection (e.g. I know this csv follows thunderstorm coventions) should be attempted, even if far from elegant. We lack file specific 'magic' although heuristics could be attempted. Simple if inelegeant interoperability is what is neded.

David-Baddeley, 2020-12-11:
Still O/T, but I'm beginning to think that the "right" way to do this is to:

- add support for io plugins for specific, easily identified (e.g. by extension) file formats. This would not address either this or, e.g. thunderstorm formatted .csv.
- add support for pluggable "translators" in the 'pipelineify' (to be renamed) recipe module. These would take the form of mappings for a mapping filter which let you map/rewrite/scale column names. (Do we also need to include default pixel scaling here?)
- have a drop-down in the csv import (and potentially other import dialogs) which let you select the csv 'dialect'

David-Baddeley, 2020-12-11:
This would let you specify a "translator" in a recipe which could be batched, and also let you switch dialects after you had loaded a file.

David-Baddeley, 2020-12-11:
Note- the translators could easily be specified as a .yaml dictionary, without requiring any actual python code.

csoeller, 2020-12-11:
Hopefully most can be captured with the mapping. I have also dealt with picasso hdfs, those need to read metadata and use that to rescale:

```
def picasso2pyme(locs,mdh):
    try:
        nmperpix = mdh['Camera.Pixelsize']
    except KeyError:
        nmperpix = mdh['Pixelsize']
        
    pyme = {}
    pyme['x'] = locs['x']*nmperpix
    pyme['y'] = locs['y']*nmperpix
    pyme['sig'] = locs['sx']*nmperpix
    pyme['nPhotons'] = locs['photons']
    pyme['error_x'] = locs['lpx']*nmperpix
    pyme['error_y'] = locs['lpy']*nmperpix
    pyme['t'] = locs['frame']
    pyme['A'] = locs['photons']/(2*math.pi)/(locs['sx']**2) # this is from the PYME kinmodels formula - check
    return pyme
```

The mdh comes from a yaml file that needs to be read. BTW, the structure of the HDF gives away it is a picasso file rather than PYME HDF.

barentine, 2020-12-11:
@David-Baddeley I think you are actually very on-topic there. That's a good plan, and points out that you don't need anything on top of what we already have for plugins (simply use a mapping filter as I mentioned above, and if remembering that if that's annoying you could write your mappings once in a recipe module in a plugin). We could add some common ones to main PYME (thunderstorm etc.)

Still separate of https://github.com/python-microscopy/pyme-plugin/issues/6, which I think would more broadly be helpful for image/localization file i/o (and for things like loading from OMERO, other servers, etc.). Notably, the equivalent of this mapping for image data (#661 sorts of things) can get gross. Other people are liable to do what we do and get a little complex in their filetypes -> does their csv include acquisition events? metadata? Can it really all be handled right with a single mapping module?

barentine, 2020-12-11:
Ah, sorry @David-Baddeley - rereading your comment now I see you are not nixing the io plugins for identifiable extensions at all, merely pointing out that to resolve degeneracy mapping modules can be used in some cases? Could still be somewhat tricky for hdf types of files, might get hacky reaching into e.g. `._h5f` of the preceding datasource to search for metadata/events/etc

David-Baddeley, 2021-12-05:
With the csv flavours now in, we should probably add a vutara flavour