python-microscopy/python-microscopy #657: Py 3.8 and 3.9 compatibility issues

David-Baddeley, 2020-12-01:
Starting as somewhere to put known issues/status on py=3.8 and 3.9. Add as encountered.

## Python==3.8

- PYMEVis appears to work 
- Some issues in dialogs (e.g. generate image) on most recent wx due to wx getting fussier. Think this also effects py3.6, 3.7
- Deprecated `time.clock()` calls need replacement (in `PYMEAcquire.mytimer`). This needs to either be fudged or wait on py2 retirement as suitable replacement calls were introduced in 3.3. PYMEAcquire does not work.
- Need to build stuff manually.

## Python==3.9

- basically no conda packages for anything (including basics such as numpy)

zacsimile, 2020-12-01:
Re: wx. We can start fixing most of those wx complaints by moving to modern wx syntax. We can do this once we drop wx3.

David-Baddeley, 2020-12-01:
The specific wx issues I was running into can and should be fixed whilst maintaining backwards compatibility (specific issue is using `wx.ALIGN_CENTER_VERTICAL` in a vertical sizer which used to be silently ignored). Think I have an uncommitted / un pushed fix on my linux computer.

barentine, 2020-12-01:
Are we not one month away from dropping wx3 support? #337

David-Baddeley, 2020-12-01:
My point was more that we should fix the specific vertical alignment problem now (as it's also likely to bite us on 3.6 and 3.7). 

As to the deadline for dropping wx3 support, it's now looking like that will slip a month or two. We might need to hold off a little on throwing the final switch, and saying we will merge breaking code, as I still haven't got the microscope machine here fully ported. This down is a combination of things, and partly the result of trying to combine the transition to py3 with a switch to win10, but the bottom line is that that has to be running happily on py3 for a month or two before I'm happy to pull the plug.

zacsimile, 2020-12-01:
I understand not wanting to switch off py2 for a couple of months, but wx3 is only supported on Mac (https://anaconda.org/david_baddeley/wxpython/files). Is this safe to drop in a month? It would be cool to start moving to the new wx syntax.

David-Baddeley, 2020-12-01:
Unfortunately not, conda is so broken on py27 at the moment that the only safe way to update is to do so in a way which doesn't require updating **any** of the dependencies.

zacsimile, 2020-12-02:
Well, we could leave `wxpython=4.0.4`, which is what we use on Windows now anyway, and start modifying functions with deprecation warnings. This would simply make the py2.7 GUI not launch on Mac, but it wouldn't involve changing any dependencies. Most of these deprecation warnings are breaking for `wxpython=4.1`, so we'd just be setting ourselves up for slowly and smoothly transitioning to modern wx by doing this.

David-Baddeley, 2020-12-02:
I think we've crossed wires here. Microscope is on wx=3, I'm not prepared to update any dependencies until we have the py3 version running

zacsimile, 2020-12-02:
Ah, OK. I thought all Windows machines were on wx4 at this point. I misunderstood

barentine, 2020-12-02:
Maybe my wires are crossed a bit too. @David-Baddeley I don't see why we need to delay dropping wx3. If you run into a regression you need fixed faster than you have time to troubleshoot at that moment (sample on scope, etc.), can't you revert to today's code, activate your py2 wx3 env and go?

David-Baddeley, 2020-12-02:
This is all rather off topic on this issue, but I'll give a concrete example of the kinds of issues I face (something I've spent far to much time on over the last couple of days)

- xy stage drivers didn't work on win10
- solution was to update the firmware
- new firmware required a complete re-write of the stage driver (for context, the old firmware used a native command set and a GCS translation DLL, the new firmware takes GCS commands directly)
- this new driver is now needed for both py2 and py3 versions of the code (because the stage firmware has been updated, the old code no longer works)

You're also aware how many open regressions there still are in the code and these do impact users of the system. The short of it is that until we are happily and reliably running on py3/win10 we still need access to the latest fixes on py2/wx3. I'm hoping to get this wrapped up soon (I'm cautiously optimistic that the stage was the last major obstacle), but with Christmas pretty much on us, this is likely to need a couple more months. 

If we absolutely need to suppress the warnings in the interim, we could think about building packages from a wx4 branch rather than master where the workflow looked something like:

1. checkout wx4 branch
2. merge master into wx4
3. build

zacsimile, 2021-05-12:
Testing on Python 3.8 (specifically 3.8.8). Steps:

- Install `numpy` from pip
- Compile `pymecompress` for Python 3.8 from `pyme-conda-recipes` and install
- `pip -r requirements.txt`
- Downgrade to `traitsui==7.1.0`

At this point PYMEVis opens and can load files.

`pymeclusterofone` and `pymeacquire` also ran, and I was able to spool a STORM simulation to the cluster. `dh5view` fails to launch when I press `Analyze`. It also fails if I try to spool to file and then try to open that file in `dh5view` after the face. In both instances, I receive the following traceback.

```
Traceback (most recent call last):
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\wx\core.py", line 3407, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "c:\users\scope_baby\code\python-microscopy\PYME\DSView\dsviewer.py", line 463, in LoadData
    vframe = DSViewFrame(im, None, im.filename, mode = mode)
  File "c:\users\scope_baby\code\python-microscopy\PYME\DSView\dsviewer.py", line 178, in __init__
    self.optionspanel = OptionsPanel(self, self.do, thresholdControls=True)
  File "c:\users\scope_baby\code\python-microscopy\PYME\DSView\DisplayOptionsPanel.py", line 128, in __init__
    vsizer.Add(self.bOptimise, 0, wx.LEFT|wx.RIGHT|wx.TOP|wx.ALIGN_CENTER|wx.EXPAND, 5)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_RIGHT | wxALIGN_CENTRE_HORIZONTAL))" failed at ..\..\src\common\sizer.cpp(2159) in wxBoxSizer::DoInsert(): Horizontal alignment flags are ignored with wxEXPAND
```

If I launch `dh5view` separately, and try to open a `.tif`, I receive the following traceback.

```
Traceback (most recent call last):
  File "C:\Users\scope_baby\miniconda3\envs\pyme\lib\site-packages\wx\core.py", line 3407, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "c:\users\scope_baby\code\python-microscopy\PYME\DSView\dsviewer.py", line 455, in LoadData
    im = ImageStack(queueURI=options.queueURI)
  File "c:\users\scope_baby\code\python-microscopy\PYME\IO\image.py", line 242, in __init__
    self.Load(filename, prompt=load_prompt, haveGUI=self.haveGUI)
  File "c:\users\scope_baby\code\python-microscopy\PYME\IO\image.py", line 1206, in Load
    self._loadTiff(filename)
  File "c:\users\scope_baby\code\python-microscopy\PYME\IO\image.py", line 873, in _loadTiff
    mdfn = self._findAndParseMetadata(filename)
  File "c:\users\scope_baby\code\python-microscopy\PYME\IO\image.py", line 715, in _findAndParseMetadata
    self.mdh.copyEntriesFrom(MetaDataHandler.XMLMDHandler(xmlfn))
  File "c:\users\scope_baby\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 759, in __init__
    self.md = self.doc.documentElement.getElementsByTagName('MetaData')[0]
IndexError: list index out of range
```

David-Baddeley, 2021-12-07:
just looking at this again ... 

## python 3.8
- no wx in conda packages for 3.8 ... (need to pip install)
- still need to pin traitsui, pyface
- display scaling issues (see #1112)

David-Baddeley, 2022-12-14:
Think we can close - running by default on 3.8 now ...

closed 2022-12-14 by David-Baddeley