python-microscopy/python-microscopy #663: [BUG] datasource slicing is pretty not numpy-like sometimes

barentine, 2020-12-02:
**Describe the bug**
```
>>> image.data.shape
[256, 256, 4, 1]
>>> a = np.ones(image.data.shape)
>>> a.shape
(256, 256, 4, 1)
>>> image.data[:,:].shape
(256, 256, 1)
>>> a[:,:].shape
(256, 256, 4, 1)
>>> image.data
<PYME.IO.DataSources.BufferedDataSource.DataSource object at 0x7fbf099c4550>
>>> image.data.dataSource
<PYME.IO.DataSources.TiffDataSource.DataSource object at 0x7fbf099c4240>
```

**additional context**
what behavior do I want for length-2 get's for #661 ?

David-Baddeley, 2020-12-03:
I'm not sure we claim that our slicing is 1:1 with numpy, just that if you slice you get a numpy array.  `image.data[:,:]` is not really well specified and it might make more sense to raise an exception if `shape[2] != 0`.

David-Baddeley, 2021-07-13:
Changed to documentation, as it's essentially a documentation issue. Has been somewhat improved with recent xyztc stuff (missing indices is an error for xyztc, and warns for legacy `image.data` access. Probably a few duplicates of this  / more general improve data model documentation type stuff.