python-microscopy/python-microscopy #664: [BUG] packages behind, cannot install on macOS 11 BigSur

barentine, 2020-12-03:
**Describe the bug**
numpy, pyopengl, etc. seem to be having some issues getting bigsur compatible packages up. When someone tried to install PYME, they got the following traceback. Opening this ticket as a place to track when:
- [ ] we support macOS 11
- [ ] we have documented the incompatibility in the install docs in the meantime

```
/Applications/python-microscopy/lib/python3.6/site-packages/PYME/misc/extraCMaps.py:73: MatplotlibDeprecationWarning: 
The revcmap function was deprecated in Matplotlib 3.2 and will be removed two minor releases later. Use Colormap.reversed() instead.
cmapdat_r = cm.revcmap(ndat[cmapname])
Traceback (most recent call last):
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/darwin.py", line 38, in GL
mode=ctypes.RTLD_GLOBAL 
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/ctypesloader.py", line 45, in loadLibrary
return dllType( name, mode )
File "/Applications/python-microscopy/lib/python3.6/ctypes/__init__.py", line 348, in __init__
self._handle = _dlopen(self._name, mode)
OSError: ('dlopen(OpenGL, 10): image not found', 'OpenGL', None)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/Applications/python-microscopy/bin/VisGUI", line 7, in 
from PYME.LMVis.VisGUI import main
File "/Applications/python-microscopy/lib/python3.6/site-packages/PYME/LMVis/VisGUI.py", line 72, in 
from PYME.LMVis import visCore
File "/Applications/python-microscopy/lib/python3.6/site-packages/PYME/LMVis/visCore.py", line 17, in 
from PYME.LMVis import gl_render3D as gl_render
File "/Applications/python-microscopy/lib/python3.6/site-packages/PYME/LMVis/gl_render3D.py", line 30, in 
from OpenGL import GLUT
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/GLUT/__init__.py", line 2, in 
from OpenGL.raw.GLUT import *
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/raw/GLUT/__init__.py", line 6, in 
from OpenGL.raw.GLUT.constants import *
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/raw/GLUT/constants.py", line 7, in 
from OpenGL import platform, arrays
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/__init__.py", line 35, in 
_load()
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/__init__.py", line 32, in _load
plugin.install(globals())
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/baseplatform.py", line 92, in install
namespace[ name ] = getattr(self,name,None)
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/baseplatform.py", line 14, in __get__
value = self.fget( obj )
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/darwin.py", line 62, in GetCurrentContext
return self.CGL.CGLGetCurrentContext 
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/baseplatform.py", line 14, in __get__
value = self.fget( obj )
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/darwin.py", line 45, in CGL
def CGL(self): return http://self.GL
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/baseplatform.py", line 14, in __get__
value = self.fget( obj )
File "/Applications/python-microscopy/lib/python3.6/site-packages/OpenGL/platform/darwin.py", line 41, in GL
raise ImportError("Unable to load OpenGL library", *err.args)
ImportError: ('Unable to load OpenGL library', 'dlopen(OpenGL, 10): image not found', 'OpenGL', None)
```

csoeller, 2020-12-03:
Hi @barentine, this stackoverflow Q+A seems to have the root cause isolated and a temporary fix until a proper solution is available: https://stackoverflow.com/questions/63475461/unable-to-import-opengl-gl-in-python-on-macos

David-Baddeley, 2020-12-03:
This looks like the fix has already made it into 3.9, and will make it into 3.8 shortly. No mention yet of backporting to any earlier versions. This suggests that we should accelerate our 3.8 attempts (see #657), which are pretty much there modulo a bit more testing and a (backwards compatible) wx fix.

David-Baddeley, 2021-03-02:
we could also try the workaround from the above linked issue which monkey-patches ctypes (potentially checking for the platform and os rather than a test import.

```python
try:
    import OpenGL as ogl
    try:
        import OpenGL.GL   # this fails in <=2020 versions of Python on OS X 11.x
    except ImportError:
        print('Drat, patching for Big Sur')
        from ctypes import util
        orig_util_find_library = util.find_library
        def new_util_find_library( name ):
            res = orig_util_find_library( name )
            if res: return res
            return '/System/Library/Frameworks/'+name+'.framework/'+name
        util.find_library = new_util_find_library
except ImportError:
    pass
```

zacsimile, 2021-03-03:
Above fix for OpenGL is verified working.

David-Baddeley, 2021-03-03:
Need to include the workaround with other entry points (PYMEAcquire, dh5view, bakeshop) and can then probably close.

David-Baddeley, 2021-04-11:
should be fixed with #916, #954

closed 2021-04-11 by David-Baddeley