python-microscopy/python-microscopy #666: [BUG] bakeshop bake not baking

barentine, 2020-12-03:
**Describe the bug**
`Bake` now appears to have no effect. no traceback either

**To Reproduce**
Steps to reproduce the behavior:
1. try and run a recipe
2. just keep trying

[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: ubuntu 16.04
 - Python version: 3.6.9
 - numpy version: 1.16
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2020-12-03:
To narrow this down I'd suggest:

- trying a much simpler recipe using only core modules (to eliminate an import issue in spawned processes - anything which imports, e.g. pylab, or pycuda could be problematic from subprocesses)
- trying on different platforms (OSX / Win) and python versions

mdlessard75, 2020-12-05:
I also get this error using no plugins (just core modules).

OS: Windows 10
Python version: 3.7.9
numpy version: 1.19.2
wxpython version [if a UI bug]: 4.0.4

barentine, 2020-12-05:
not a simpler recipe, but I am again hitting this on 
OS: mac OSX 10.15
Python version: 3.7.7
numpy version: 1.19.2
wxpython version [if a UI bug]: 4.0.4

barentine, 2020-12-09:
I reverted the batch code to just call pool map rather than map_async - I wonder if the error it raises is related to what we get with the current code (seeing as how the difference is potentially not much more than who calls map_async)
```
Error whilst running Batch Analysis
==============================================
python-microscopy=20.12.03
python=3.7.7, platform=darwin
numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/recipeGui.py", line 821, in OnBake
    batchProcess.bake(self.rm.activeRecipe, {'input':self.inputFiles}, out_dir, num_procs=num_procs)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/batchProcess.py", line 91, in bake
    pool.map(runRec, taskParams)
  File "/opt/miniconda3/lib/python3.7/multiprocessing/pool.py", line 268, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
  File "/opt/miniconda3/lib/python3.7/multiprocessing/pool.py", line 657, in get
    raise self._value
  File "/opt/miniconda3/lib/python3.7/multiprocessing/pool.py", line 431, in _handle_tasks
    put(task)
  File "/opt/miniconda3/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/opt/miniconda3/lib/python3.7/multiprocessing/reduction.py", line 51, in dumps
    cls(buf, protocol).dump(obj)
TypeError: can't pickle WeakMethod objects

```

barentine, 2021-03-02:
I think we don't see that error because we wait(). See [here](https://stackoverflow.com/questions/50572875/python-pool-map-async-doesnt-wait-for-wait).

```
Error whilst running Batch Analysis
==============================================
python-microscopy=21.03.01
python=3.6.9, platform=linux
numpy=1.16.5, wx=4.0.4 gtk2 (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/recipes/recipeGui.py", line 877, in OnBake
    batchProcess.bake(self.rm.activeRecipe, {'input':self.inputFiles}, out_dir, num_procs=num_procs)
  File "/home/smeagol/code/python-microscopy/PYME/recipes/batchProcess.py", line 87, in bake
    pool.map(runRec, taskParams)
  File "/home/smeagol/miniconda3/lib/python3.6/multiprocessing/pool.py", line 266, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
  File "/home/smeagol/miniconda3/lib/python3.6/multiprocessing/pool.py", line 644, in get
    raise self._value
  File "/home/smeagol/miniconda3/lib/python3.6/multiprocessing/pool.py", line 424, in _handle_tasks
    put(task)
  File "/home/smeagol/miniconda3/lib/python3.6/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/home/smeagol/miniconda3/lib/python3.6/multiprocessing/reduction.py", line 51, in dumps
    cls(buf, protocol).dump(obj)
TypeError: can't pickle WeakMethod objects

```

barentine, 2021-03-02:
```
In [1]: from PYME.recipes.base import ModuleCollection

In [2]: rec = ModuleCollection()

In [3]: import pickle

In [4]: pickled = pickle.dumps(rec)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-4-26c2908e6393> in <module>
----> 1 pickled = pickle.dumps(rec)

TypeError: can't pickle _thread.lock objects
```

David-Baddeley, 2021-03-02:
This points towards a pickling error with the recipe objects.

I wonder if it is a) py3 related or b) relates to locks we have added somewhere. It might be safer to pass the recipe to child processes as text than as a recipe object.

David-Baddeley, 2021-03-02:
OK, so 2 issues:

1) errors are silenced
2) pickling error

Should hopefully have fixes soon ...

David-Baddeley, 2021-03-02:
Fixed locally, will commit and push later this evening.

barentine, 2021-03-02:
Thanks @David-Baddeley!

closed 2021-04-11 by David-Baddeley