python-microscopy/python-microscopy #669: [BUG] Pos: slider not responsive using arrow keys or intermittently responsive using mouse

mdlessard75, 2020-12-04:
After loading a multi-dimensional image (3D, multi-color, etc.) the viewing window intermittently updates the z position/channel while clicking and moving the Pos: slider. Selecting the slider and using the right and left arrow keys results in no change/update of the viewing window.

**Expected behavior**
I expect that moving the Pos: slider with the mouse or with the arrow keys will result in an updated image representing the selected z plane or color channel.

**Platform**
 - OS: Windows 10
 - Python version: 3.7.9
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

zacsimile, 2020-12-04:
Gave this a quick go binding the mouse keys. For some reason `self.Bind(wx.EVT_KEY_DOWN, self.OnKeyPress)` doesn't seem to bind key pressing in `PYME.DSView.modules.playback.PlayPanel`. Is this superseded somewhere @David-Baddeley?

David-Baddeley, 2020-12-06:
Behaviour of arrow keys will be platform dependant, and shouldn't rely on explicit key bindings. On platforms where sliders do respond to key strokes (Win, linux) it should just do the right thing without any specific binding. IE - you shouldn't need to bind the keys (this probably also explains why explicitly binding them doesn't do anything - they are already consumed by the underlying control and don't propagate up to stack to where you are binding them).

Having a quick look this looks like a refresh/update issue. At some point I reduced the amount of refreshes in the viewer to improve performance (display was being refreshed multiple times and it was causing flicker / etc ...), suspect I might have taken it too far. Reproducible on OSX and Linux, py2.7 and py3.7.

For reference, pgdown/pgup and mouse wheel will also both scroll through z when the image has focus.

David-Baddeley, 2020-12-06:
Actually, just had a look at the code, this is probably due to the fact that we bind to `EVT_SCROLL_THUMBRELEASE` which means that we only actually change the position when we stop dragging. This is a deliberate performance optimisation to allow you to move quickly through a large series, avoiding drawing lots of intermediate frames. I think this behaviour (only update position at end of scroll) is desirable for snappy behaviour, particularly as we load large blinking series lazily and loading an intermediate frame involves reading data from disk, decompressing, and potentially a large amount of computation if we have background subtraction turned on. When we were updating on every `EVT_SCROLL`, it was quite common to get a huge backlog of updates so that the display kept changing for several seconds or tens of seconds after you'd stopped scrolling as it worked through all the updates. With a large series this could effectively leave the program unresponsive.

That said, being able to use the keyboard cursors to 'bump' the slider would be desirable (I've always just used pgup/pgdown for this, but it's much less discoverable).

An ideal solution would:
- update for key-based changes ("bumping")
- not create a flood of updates whilst sliding
- potentially still update during sliding, but in a throttled way (this will be hard to get right)

David-Baddeley, 2020-12-06:
I suspect that the arrow keys end up generating `EVT_SCROLL_LINEUP` and `EVT_SCROLL_LINEDOWN` - we should look at processing these as well. We just can't afford to bind `EVT_SCROLL`, `EVT_SCROLL_THUMBTRACK`, or (probably) `EVT_SCROLL_CHANGED`.

See https://wxpython.org/Phoenix/docs/html/wx.ScrollEvent.html

David-Baddeley, 2020-12-06:
Binding `EVT_SCROLL_LINEUP` and `EVT_SCROLL_LINEDOWN` seems to fix the keyboard scrolling on linux. Patch soon.

David-Baddeley, 2020-12-06:
Minimal fix in #676. Note that this leaves dragging behaviour as is (only updates when you release the slider).

### TODO:

- is there any need to make updates as you slide?
- should we also bind `EVT_SCROLL_PAGEDOWN` and `EVT_SCROLL_PAGEUP`? If we do this, can / should we also make pgdown/pgup make one frame updates (as they do in the main viewer) rather than the default 10 frame updates.

David-Baddeley, 2021-02-24:
going to call this fixed - can reopen if needed

closed 2021-02-24 by David-Baddeley