python-microscopy/python-microscopy #671: [BUG] z dwell time checked when checking if we want to do a preflight, even for non-z protocols

barentine, 2020-12-04:
**Describe the bug**
```
(10.160710895153072, 'spoolController.StartSpooling', {'maxFrames': 10, 'protocol': 'htsms-widefield.py'}, 1.7976931348623157e+308, 200.0), spoolController.StartSpooling
DEBUG:PYME.Acquire.SpoolController:Looking for Bergamot/2020_12_4/PKfirst8wellibidi/000/4_12_series_D (.pcs or .h5) on cluster

Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ActionManager.py", line 140, in Tick
    self.isLastTaskDone = fcn(**args)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\SpoolController.py", line 415, in StartSpooling
    if doPreflightCheck and not preflight.ShowPreflightResults(None, protocol.PreflightCheck()):
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\protocol.py", line 117, in PreflightCheck
    if not eval(c.check):
  File "<string>", line 1, in <module>
AttributeError: 'TaskListProtocol' object has no attribute 'dwellTime'
```

note that spool controller is set to 'Standard' not 'Z stepped'

**Additional context**
will look at this more in a bit. are we somehow getting the default preflight from the z task protocol even though the normal protocol is active?

barentine, 2020-12-04:
note that either way, for recipes.acquisitions queuing we probably aught to default doPreflight to false

barentine, 2020-12-04:
I'm a little frustrated that said default preflight task now exists. Ran headlong into it with a multiwell sample, and think that if someone is writing protocols they will similarly not appreciate being babysat like that. I know, timing is tough below 0.1 s, but I'm not trying to take a nice z stack (or any z stack at the moment) I'm just trying to debug my workflow, and I wasn't asking for the trainwreck

barentine, 2020-12-04:
could be that using a default parameter as a mutable object (preflightList = []) is what's hitting us here

David-Baddeley, 2021-01-28:
hmm - it's possible, but shouldn't be

barentine, 2021-02-11:
still fun times with this
```
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ActionManager.py", line 200, in Tick
    self.isLastTaskDone = action(self.scope())
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\actions.py", line 85, in __call__
    return self._do_then(scope)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\actions.py", line 62, in _do_then
    return self._then(scope)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\actions.py", line 97, in __call__
    return scope.spoolController.StartSpooling(**self._args)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\SpoolController.py", line 417, in StartSpooling
    if doPreflightCheck and not preflight.ShowPreflightResults(None, protocol.PreflightCheck()):
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\protocol.py", line 117, in PreflightCheck
    if not eval(c.check):
  File "<string>", line 1, in <module>
AttributeError: 'TaskListProtocol' object has no attribute 'dwellTime'
```

barentine, 2021-02-23:
you can hit this in the simulator if you feel like it, running simul488.py protocol
```
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/Acquire/ui/HDFSpoolFrame.py", line 475, in OnBStartSpoolButton
    self.spoolController.StartSpooling(fn, #stack=stack, #compLevel = compLevel,
  File "/Users/Andrew/code/python-microscopy/PYME/Acquire/SpoolController.py", line 417, in StartSpooling
    if doPreflightCheck and not preflight.ShowPreflightResults(None, protocol.PreflightCheck()):
  File "/Users/Andrew/code/python-microscopy/PYME/Acquire/protocol.py", line 117, in PreflightCheck
    if not eval(c.check):
  File "<string>", line 1, in <module>
AttributeError: 'TaskListProtocol' object has no attribute 'dwellTime'
```

ecourchaine, 2021-03-09:
This caused some pretty big problems today while trying to benchmark some automatic roi finding.  I wanted a "standard" widefield snapshot at each of some 200 rois and the preflight's all stacked on top of each other and prevented a queue pause.

David-Baddeley, 2021-03-09:
Will have a look, as it really shouldn't be effecting the non-z stacked acquisitions. 

You certainly can (and potentially should, as it requires user intervention) set `doPreflightCheck` to `False` when queueing acquisitions in the action manager. I do think it's better to have this as an explicit opt-out, rather than a default though. Suspect solution has 2 parts:

1) find why the dwellTime check is being run for non z-stack acquisitions
2) work out a sensible way of handling preflight checks for automated acquisition. I think there are a number of options here, but displaying a GUI dialog and stalling is clearly wrong. Options:
    - don't run preflights for automated acquisitions (a bit risky for my taste, but a potential short term workaround, and available now)
    - change the checks to log a warning and keep going when used in automated mode (current GUI checks is essentially just a warning, although you get a chance to abort). This would be the most permissive option - basically you can still get crap data, but at least there will be breadcrumbs as to why.
    - change checks to raise an exception and abort acquisition (probably too harsh)
    - extend the checks so that you can specify if the check is strict or (throws and exception) or not (just logs a warning)

David-Baddeley, 2021-04-11:
This is partially fixed as of #799 and #902 in that you now have an option of  / default to soft failures on preflights when queued through the action manager.

PR #1001 (2021-04-12): Issue 671
Fixes #671, turns out it was the mutable default, but not entirely clear why.
closed 2021-04-12 by David-Baddeley