python-microscopy/python-microscopy #673: [BUG] bioformats only works for first load (ie fails in batch processing)

barentine, 2020-12-05:
**Describe the bug**
A clear and very-unconcise description: we can't open most multichannel tif's coming out of Fiji, which is frustrating. One goes through the effort to install python-bioformats (we knowingly are leaving the install instructions broken - one has to ping Zach for what works [`export CFLAGS='-Wno-implicit-function-declaration'` followed by `pip install python-javabridge`], and once you finally get there, change the extension of all your tif's to tiff to trigger bioformats instead of our internal tif loading, build/debug your recipe, you go to batch it and find that we don't properly handle cleanup so you can only run it on one input at a time

<details> <summary> traceback </summary>

```
(base) Andrews-MacBook-Pro-5:~ Andrew$ bakeshop
Trying to load 3rd party recipe module pyme_omero.recipe_modules.omero_upload
Error loading plugin: pyme_omero.recipe_modules.omero_upload
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/modules.py", line 41, in <module>
    m = __import__(mn, fromlist=mn.split('.')[:-1])
ModuleNotFoundError: No module named 'pyme_omero'
Trying to load 3rd party recipe module quant_condensate.recipe_modules.cluster_quant
Loaded 3rd party recipe module quant_condensate.recipe_modules.cluster_quant
DEBUG:PYME.recipes.recipeGui:BatchFrame.__init__ start
DEBUG:PYME.recipes.recipeGui:BatchFrame.__init__ done
/Users/Andrew/code/python-microscopy/PYME/ui/progress.py:82: wxPyDeprecationWarning: Using deprecated class. Use Cursor instead.
  self.window.SetCursor(wx.StockCursor(wx.CURSOR_WAIT))
DEBUG:matplotlib.pyplot:Loaded backend SVG version unknown.
filename == /Users/Andrew/Documents/BewersdorfLab/STED_CBs_stacks/Coilin594_SMN647_MS023_001.tiff
Bioformats:loading data
DEBUG:javabridge.jutil:Creating JVM object
DEBUG:javabridge.jutil:Launching VM in non-python thread
DEBUG:javabridge.jutil:Attaching to VM in monitor thread
DEBUG:javabridge.jutil:Signalling caller
12:07:51.647 [Thread-1] DEBUG loci.common.NIOByteBufferProvider - Using mapped byte buffer? false
12:07:51.672 [Thread-1] DEBUG loci.formats.ClassList - Could not find loci.formats.in.URLReader
java.lang.ClassNotFoundException: loci.formats.in.URLReader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_65]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_65]
	at java.lang.Class.forName0(Native Method) ~[na:1.8.0_65]
	at java.lang.Class.forName(Class.java:264) ~[na:1.8.0_65]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.<init>(ClassList.java:138) [bioformats_package.jar:6.5.1]
12:07:51.675 [Thread-1] DEBUG loci.formats.ClassList - Could not find loci.formats.in.SlideBook6Reader
java.lang.ClassNotFoundException: loci.formats.in.SlideBook6Reader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_65]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_65]
	at java.lang.Class.forName0(Native Method) ~[na:1.8.0_65]
	at java.lang.Class.forName(Class.java:264) ~[na:1.8.0_65]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.<init>(ClassList.java:138) [bioformats_package.jar:6.5.1]
12:07:51.676 [Thread-1] DEBUG loci.formats.ClassList - Could not find loci.formats.in.ScreenReader
java.lang.ClassNotFoundException: loci.formats.in.ScreenReader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_65]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_65]
	at java.lang.Class.forName0(Native Method) ~[na:1.8.0_65]
	at java.lang.Class.forName(Class.java:264) ~[na:1.8.0_65]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.<init>(ClassList.java:138) [bioformats_package.jar:6.5.1]
12:07:52.006 [Thread-1] DEBUG loci.formats.ClassList - Could not find loci.formats.in.URLReader
java.lang.ClassNotFoundException: loci.formats.in.URLReader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_65]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_65]
	at java.lang.Class.forName0(Native Method) [na:1.8.0_65]
	at java.lang.Class.forName(Class.java:264) [na:1.8.0_65]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:6.5.1]
[... 256 lines cut ...]
    bioformats_file = BioformatsDataSource.BioformatsFile(filename)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/DataSources/BioformatsDataSource.py", line 75, in __init__
    super(BioformatsFile, self).__init__(*args, **kwargs)
  File "/opt/miniconda3/lib/python3.7/site-packages/bioformats/formatreader.py", line 626, in __init__
    self.path)
  File "/opt/miniconda3/lib/python3.7/site-packages/javabridge/jutil.py", line 1717, in make_instance
    klass = get_env().find_class(class_name)
AttributeError: 'NoneType' object has no attribute 'find_class'

```

</details>

**To Reproduce**
Steps to reproduce the behavior:
1. open bakeshop
2. batch recipe on multiple files which need to be loaded by bioformats

note this occurs of course when running batchProcess from scripts as well

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: mac OSX 10.15
 - Python version: 3.7.7
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

zacsimile, 2020-12-05:
I'm not 100% sure this a cleanup issue (although it might be). The problem outlined in #277 keeps `dh5view` open after closing the image because we maintain a reference to the image, which means the `BioformatsFile` `__del__()` method never gets called, so the VM is never released. If we're batching multiple files, it should be OK that the VM stays on in the background. Each new file will just add a VM reference count.

zacsimile, 2020-12-05:
This appears to be the inverse problem, the vm is killed successfully but not restarted successfully.

barentine, 2020-12-05:
currently bypassing this cleanup/restart by ignoring del on the `BioformatsDataSource.BioformatsFile`
[image]

thanks for the help @zacsimile !

zacsimile, 2020-12-05:
Two possible things going on:

- The `numVMRefs` counting is off and `javabridge.start_vm()` doesn't get called on the second file.
- `javabridge.start_vm()` does get called and there's some sort of timeout preventing the JVM restart

David-Baddeley, 2020-12-05:
Option 3,

You can't open the vm twice due to bugs in either java or javabridge.

zacsimile, 2020-12-05:
@David-Baddeley, sorry, I edited in option 3 and an explanation for why this occurring after I posted. I have a bad habit of refining my thoughts on GitHub after posting. Don't trust any message you receive from me via GitHub email.

zacsimile, 2020-12-05:
I do think option 3 is what's happening. We should probably make `release_VM()` blocking and also create a flag that we can use in bakeshop that says “don’t kill the VM”.

David-Baddeley, 2020-12-05:
I don't think any of this should be multi-threaded (so probably not a blocking/ race condition issue). Ultimately it might make sense to just open a VM and re-use if we could find a way to ensure it got cleaned up at exit. I also have a feeling that this used to work.

For the motivating problem, the right solution is to either a) use the bioformats exporter rather than save in ImageJ so you get sensibly formatted TIFF or b) fix our import code (ideally both - philosophically we should be able to import ImageJ TIFFs, but seeing as ImageJ tiff handling is pretty broken and a moving target, having (option a) as a documented fallback is a good idea).

zacsimile, 2020-12-05:
I believe the VM is in a different thread and it takes it longer to shut down than it takes us to hit the next iteration of the loop. As such, the VM looks still live when we go to open the next `BioformatsFile`, but is in fact in the process of turning off.

zacsimile, 2020-12-06:
Btw, we do in theory open a single VM and re-use it. I think in this batch case we close the `BioformatsFile` after we're done with it, and before opening the next file, and this causes us to release the java VM as the number of references (1) drops to zero. We just need to set up the reference counting a bit different in the batch files or open the next file before closing the previous file.

David-Baddeley, 2020-12-06:
I wonder if it's better to just keep the VM open for the duration of the process, if we can find a way of making sure that it gets destroyed when the rest of the process dies (can we hook something to tell us when the main thread exits and/or make the VM thread a deamon thread)?

barentine, 2020-12-06:
For what it's worth the javabridge.kill_vm docstring says once it's killed it can't be restarted, so yes I think we're probably stuck keeping the VM open until the process is killed, otherwise we're liable to hit it later.

zacsimile, 2020-12-06:
@David-Baddeley we'd discussed this a bit in #293.

@barentine nice catch.

David-Baddeley, 2020-12-06:
Maybe something along these lines:

```
haveVM = False

def ensure_VM():
    global haveVM
    
    if not haveVM:
        javabridge.start_vm(class_path=bioformats.JARS, run_headless=True)
        haveVM = True
        atexit.register(javabridge.kill_vm)
    
def release_VM():
    pass
```

zacsimile, 2020-12-06:
Won't that create the same problem you were concerned about in #293? If we have two instances of `dh5view` open and we close one, the VM used by both will be closed.

David-Baddeley, 2020-12-06:
nope - `atexit()` should be global for the interpreter.

zacsimile, 2020-12-06:
Well that's nifty. That should solve #277 as well.

David-Baddeley, 2020-12-06:
ie should be triggered when we drop out of `app.MainLoop()`

David-Baddeley, 2020-12-06:
with any luck ...

zacsimile, 2020-12-06:
Nope, doesn't fix #277. Might fix Andrew's error, though.

David-Baddeley, 2020-12-06:
What happens if you put `sys.exit()` after the Main Loop call in dsviewer?

zacsimile, 2020-12-06:
Also does not work

zacsimile, 2021-05-18:
@barentine did David's solution fix your error?