python-microscopy/python-microscopy #678: [BUG] PYMEAcquire fails in Linux with wxPython>=4.1.0 (wxWidgets>=3.1)

carandraug, 2020-12-06:
PYMEAcquire is incompatible with wxPython 4.1.0 release, at least on Linux and GTK.

**To Reproduce**

Steps to reproduce the behaviour:
1. Start `PYMEAcquire` from command line (so it picks the defaul `init.py` with demo devices)
2. Note that it won't start and the following erros:

```
ERROR:PYME.Acquire.acquiremainframe:Error whilst initializing Simulation UI GUI
Traceback (most recent call last):
  File "/home/carandraug/src/PyME/PYME/Acquire/acquiremainframe.py", line 238, in doPostInit
    cm.run(self, self.scope)
  File "/home/carandraug/src/PyME/PYME/Acquire/ExecTools.py", line 205, in run
    self.codeObj(parent, scope)
  File "/home/carandraug/src/PyME/PYME/Acquire/Scripts/init.py", line 92, in sim_controls
    dsc = simui_wx.dSimControl(MainFrame, scope.simcontrol)
  File "/home/carandraug/src/PyME/PYME/Acquire/Hardware/Simulator/simui_wx.py", line 425, in __init__
    self._init_ctrls(parent)
  File "/home/carandraug/src/PyME/PYME/Acquire/Hardware/Simulator/simui_wx.py", line 216, in _init_ctrls
    2)
wx._core.wxAssertionError: C++ assertion "!(flags & wxALIGN_CENTRE_HORIZONTAL)" failed at /tmp/pip-install-2004mlpb/wxpython/ext/wxWidgets/src/common/sizer.cpp(2106) in DoInsert(): Horizontal alignment flags are ignored in horizontal sizers
[...]
ERROR:PYME.Acquire.acquiremainframe:Error whilst initializing Camera controls GUI
Traceback (most recent call last):
  File "/home/carandraug/src/PyME/PYME/Acquire/acquiremainframe.py", line 238, in doPostInit
    cm.run(self, self.scope)
  File "/home/carandraug/src/PyME/PYME/Acquire/ExecTools.py", line 205, in run
    self.codeObj(parent, scope)
  File "/home/carandraug/src/PyME/PYME/Acquire/Scripts/init.py", line 100, in cam_controls
    scope.camControls['Fake Camera'] = AndorControlFrame.AndorPanel(MainFrame, scope.cam, scope)
  File "/home/carandraug/src/PyME/PYME/Acquire/Hardware/AndorIXon/AndorControlFrame.py", line 203, in __init__
    self._init_ctrls()
  File "/home/carandraug/src/PyME/PYME/Acquire/Hardware/AndorIXon/AndorControlFrame.py", line 151, in _init_ctrls
    sbCooling.Add(hsizer2, 0, wx.EXPAND|wx.ALIGN_CENTER_HORIZONTAL, 0)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_RIGHT | wxALIGN_CENTRE_HORIZONTAL))" failed at /tmp/pip-install-2004mlpb/wxpython/ext/wxWidgets/src/common/sizer.cpp(2089) in DoInsert(): Horizontal alignment flags are ignored with wxEXPAND
[...]
ERROR:PYME.Acquire.acquiremainframe:Error whilst initializing Laser controls GUI
Traceback (most recent call last):
  File "/home/carandraug/src/PyME/PYME/Acquire/acquiremainframe.py", line 238, in doPostInit
    cm.run(self, self.scope)
  File "/home/carandraug/src/PyME/PYME/Acquire/ExecTools.py", line 205, in run
    self.codeObj(parent, scope)
  File "/home/carandraug/src/PyME/PYME/Acquire/Scripts/init.py", line 157, in laser_controls
    lsf = lasersliders.LaserSliders(MainFrame.toolPanel, scope.state)
  File "/home/carandraug/src/PyME/PYME/Acquire/ui/lasersliders.py", line 88, in __init__
    sz.Add(sl, 1, wx.ALL|wx.ALIGN_CENTER_VERTICAL|wx.EXPAND, 2)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL))" failed at /tmp/pip-install-2004mlpb/wxpython/ext/wxWidgets/src/common/sizer.cpp(2114) in DoInsert(): Vertical alignment flags are ignored with wxEXPAND
[...]
Traceback (most recent call last):
  File "/home/carandraug/src/PyME/PYME/ui/mytimer.py", line 42, in Notify
    a()
  File "/home/carandraug/src/PyME/PYME/Acquire/acquiremainframe.py", line 170, in _check_init_done
    self.doPostInit()
  File "/home/carandraug/src/PyME/PYME/Acquire/acquiremainframe.py", line 252, in doPostInit
    self._add_live_view()
  File "/home/carandraug/src/PyME/PYME/Acquire/acquiremainframe.py", line 200, in _add_live_view
    self.vsp = disppanel.dispSettingsPanel2(self, self.vp)
  File "/home/carandraug/src/PyME/PYME/DSView/displaySettingsPanel.py", line 175, in __init__
    hsizer.Add(self.hlDispMapping, 1, wx.EXPAND|wx.ALIGN_CENTER_VERTICAL|wx.ALL, 5)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL))" failed at /tmp/pip-install-2004mlpb/wxpython/ext/wxWidgets/src/common/sizer.cpp(2114) in DoInsert(): Vertical alignment flags are ignored with wxEXPAND
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Debian 10
 - Python version: 3.7.3
 - wxpython version: 4.1.0 (same problem with 4.1.1, no problem with 4.0.7) 

**Additional context**

It works fine with wxPython 4.0.7, the issue is wxPython which with version 4.1.0 started to build against wxWidgets development releases (3.1 series) instead of stable releases (the 3.0 series). It is also builds wxWidgets with debug so these new errors are C++ assertions which were present before but were being ignored and now raise python exceptions.

I can give it a go at fixing this (I fixed the same recently for cockpit) but I was under the impression you favour conda and pinning versions. Let me know.

David-Baddeley, 2020-12-06:
Many thanks for the report. We currently have a bit of a funny status with wx, in that we have a couple of production systems on py27 / wx3 that, for various reasons, we haven't been able to update to py3.x/wx4 yet. Because of this we don't want to merge changes into master which break wx3 compatibility.

We can break wx4 problems into 3 classes:

1.  was wrong on wx3 (if silently ignored), and is still wrong on wx4. The particular error you have struck (`wx.ALIGN_CENTRE_VERTICAL` in a vertical `BoxSizer`) falls into this category (this particular error is repeated in multiple different places in the code - it also occurs, e.g. in the PYMEVis image generation dialog).
2. OK on wx3, broken on wx4, but with a simple backwards compatible fix
3. Correct on wx3, deprecated in wx4.0.x, broken in wx4.1.x with no backwards compatible fix (e.g. `wx.EmptyBitmap()` -> `wx.Bitmap()`)

Categories 1 & 2 should be fixed immediately (PRs welcome), and we're collecting but not yet merging PRs for category 3 (e.g. #207). Our initial deadline for having the remaining systems on wx4 and merging category 3 changes was going to be this coming Jan, but having lost around 3 months of lab access to covid, this is likely to slip a few months. Because we're now recommending wx4 for new installs (and it's getting hard to get 4.0.x on, e.g. py=3.8) we might create a wx4 branch to collect all the category 3 changes in the interim and use this for package builds etc ...

To summarise, a PR would be welcome. If it just touches the sizer alignment stuff it will likely be merged quickly, if it also fixes some of the other wx4 changes (e.g. bitmaps) it will likely sit for a while before being merged.

carandraug, 2020-12-07:
> To summarise, a PR would be welcome. If it just touches the sizer alignment stuff it will likely be merged quickly, if it also fixes some of the other wx4 changes (e.g. bitmaps) it will likely sit for a while before being merged.

PR made. I understand the not fixing the Bitmaps stuff, those are only warnings and one can live with those.

David-Baddeley, 2020-12-07:
Merged. Can probably close this, but similar issues exist in other parts of code.

closed 2020-12-07 by David-Baddeley