python-microscopy/python-microscopy #681: [BUG] Get segmented IDs from image doesn't work if the generating points don't have voxeldata metadata

zacsimile, 2020-12-07:
**Describe the bug**
If I open a localization file sans metadata (e.g. a .csv or .mat), generate a rendering, threshold and label, and then try to pull the labels back into the localizations, I get the following traceback.

```
 Error whilst running Analysis>Get segmented IDs from image
==============================================
python-microscopy=20.12.07
python=3.6.12, platform=win32
numpy=1.14.6, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\IO\MetaDataHandler.py", line 285, in __getitem__
    return self.getEntry(name)
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\IO\MetaDataHandler.py", line 547, in getEntry
    return eval('self.'+entryName)
  File "<string>", line 1, in <module>
AttributeError: 'NestedClassMDHandler' object has no attribute 'voxelsize'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\ui\progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\LMVis\Extras\objectMeasurements.py", line 75, in OnGetIDs
    recipe.add_modules_and_execute([mod,])
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\recipes\base.py", line 921, in add_modules_and_execute
    self.execute()
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\recipes\base.py", line 665, in execute
    m.execute(self.namespace)
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\recipes\localisations.py", line 600, in execute
    ids, numPerObject = cluster_morphology.get_labels_from_image(img, inp, minimum_localizations=self.minimum_localizations)
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\Analysis\points\cluster_morphology.py", line 46, in get_labels_from_image
    pixX, pixY, pixZ = pixel_index_of_points_in_image(label_image, points)
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\Analysis\points\coordinate_tools.py", line 223, in pixel_index_of_points_in_image
    p_ox = roi_x0 * points.mdh.voxelsize_nm.x
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\IO\MetaDataHandler.py", line 416, in voxelsize_nm
    return get_voxelsize_nm(self)
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\IO\MetaDataHandler.py", line 223, in get_voxelsize_nm
    return VoxelSize(1e3*mdh['voxelsize.x'], 1e3*mdh['voxelsize.y'], 1e3*mdh.get('voxelsize.z', 0))
  File "C:\Users\zrc4\python-microscopy\lib\site-packages\PYME\IO\MetaDataHandler.py", line 287, in __getitem__
    raise KeyError('Key %s not defined' % name)
KeyError: 'Key voxelsize.x not defined'
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10/macOS Catalina
 - Python version: 3.6.12/3.7.9
 - numpy version: 1.14.6/1.19.12
 - wxpython version [if a UI bug]: 4.0.4

zacsimile, 2020-12-07:
Two ways I can think of to fix this, and I'm not sure which is preferable.

- Change `PYME.IO.MetaDataHandler.get_voxelsize_nm()` to return a default of 0 if `x` or `y` voxelsize isn't found. This is what we already do for the `z` voxelsize.
- Change `PYME.Analysis.points.coordinate_tools.pixel_index_of_points_in_image()` to return a default of 0 for `voxelsize_nm.x` and `y` in the `try` block.

zacsimile, 2020-12-07:
e4\

David-Baddeley, 2020-12-07:
Hmm, it should work as we should add the voxelsize metadata to the
generated images regardless of whether there is metadata in the original
dataset.

Critically, the voxelsize should be that of the generated image, not the
raw data.

zacsimile, 2020-12-07:
Yes, but `PYME.Analysis.points.coordinate_tools.pixel_index_of_points_in_image()` uses the voxelsize of the original image.

closed 2020-12-07 by zacsimile