python-microscopy/python-microscopy #682: [BUG] PYMEVis File>Open not displaying HDF files when All supported formats is selected

zacsimile, 2020-12-07:
**Describe the bug**
PYMEVis File>Open not displaying HDF files when All supported formats is selected. `csv`/`txt`/`mat` all show up.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10
 - Python version: 3.6.12
 - numpy version: 1.14.6
 - wxpython version [if a UI bug]: 4.0.4

zacsimile, 2020-12-07:
This one's fairly confusing. 3/5 file types work. The other two work when selected individually. The file types are all passed as expected according to https://wxpython.org/Phoenix/docs/html/wx.FileDialog.html.

David-Baddeley, 2020-12-07:
Not tested, but I strongly suspect it's the brackets in the extension specification for "All supported formats"

i.e. should be 
```python
filename = wx.FileSelector("Choose a file to open", 
                                   nameUtils.genResultDirectoryPath(), 
                                   wildcard='All supported formats|*.h5r;*.txt;*.mat;*.csv;*.hdf|PYME Results Files (*.h5r)|*.h5r|Tab Formatted Text (*.txt)|*.txt|Matlab data (*.mat)|*.mat|Comma separated values (*.csv)|*.csv|HDF Tabular (*.hdf)|*.hdf')
```

otherwise you are trying to match `'(*.h5r'` and `'*.hdf)'`.

zacsimile, 2020-12-07:
Oh good lord, yes that's it.

closed 2020-12-07 by zacsimile