python-microscopy/python-microscopy #690: [BUG] dtype warning in H5Exporter a little overzealous?

barentine, 2020-12-08:
**Describe the bug**
Would be nice when saving an h5 from a recipe, not to fear the following warning
```
/Users/Andrew/code/python-microscopy/PYME/IO/dataExporter.py:588: UserWarning: Attempting to save an unsupported data-type (float64) - data should be one of uint8, uint16, or float32
  exp.Export(ds, filename, slice(0, ds.shape[0], 1), slice(0, ds.shape[1], 1), slice(0, ds.shape[2], 1),mdh, events, origName, progressCallback=progressCallback)

```

when it doesn't appear to be an issue
```
<PYME.IO.DataSources.BGSDataSource.DataSource object at 0x7fb81d1d9cd0>
>>> image.data.datasource
<PYME.IO.DataSources.HDFDataSource.DataSource object at 0x7fb81c749ed0>
>>> image.data.datasource.getSlice(0)
array([[ 534.08274395,  900.20846307, 1277.37740114, ..., 1222.41806856,
        1257.50020103, 1126.76262905],
       [ 103.64516554, 1096.96820454, 1403.43696462, ..., 1795.99117118,
        1148.39232441, 1517.81138828],
       [ 249.59724522, 1286.90823314,  123.26793198, ..., 2175.37725094,
        1477.47122482, 1697.39423224],
       ...,
       [ 101.2696174 ,   53.63074939,  108.62124473, ...,   73.16046848,
         185.64493098,  124.84680914],
       [ 199.00332612,  118.06676926,  131.18297886, ...,  165.3252766 ,
         364.44462168,  143.21197101],
       [ 212.07474454,  245.97458754,  130.65662876, ...,  255.05696272,
         776.25907601,  684.22927208]])
>>> image.data.datasource.getSlice(0).dtype
dtype('float64')
```

**To Reproduce**
Steps to reproduce the behavior:
1. run some analysis which produces a float64 (or int32, or int64) datasource underlying an ImageStack
2. ImageOutput module save to h5
3. See warning
4. panic

don't warn if it isn't a problem - I think the tables get atom for dtype raises if it has issues, maybe we should rely on that?

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: osx 10.15
 - Python version: 3.7.7
 - numpy version: 1.19
 - wxpython version [if a UI bug]: 4.0.4

**Additional context**
- pytables 3.6.1

Notably, if it _is_ an issue, it would at least be nice to make the warning more descriptive / tell us what we do (do we just try and see if pytables can take it? do we down-grade float64 to float32? do we do something dangerous with a signed int to an unsigned int? I asked all sorts of questions in a short period of time it took to open the code)

David-Baddeley, 2020-12-08:
Have you tried opening the generated image in dh5view/PYMEImage? There's nothing specific about HDF which prevents you from saving as different formats, but the data might not load correctly and/or you might get truncation and rounding issues if you open, process and save. There was one point in time where the PYMEImage viewer would segfault on float64 data (I think this is fixed, but maybe not).

The thoughts behind the warning are as follows: 
- dh5view/PYMEImage has LUT code for the 3 data types mentioned
- things are a bit more predictable if we restrict our image types. 
- at least some of the recipe modules (definitely deconvolution, but also a lot of stuff like background correction) operate on float32, meaning that you might get silent precision loss with float64.
- The uint8, uint16, float32 types are those supported by the TIFF format and PZF formats (i.e. if you have a workflow which requires e.g. float64 it's not going to be super portable.
- Those data types should be sufficient for most if not all image processing tasks and keep memory footprints reasonable.

In theory, it's an error for a recipe module to produce an image which is not one of those data types (although this is not well enforced or documented, and violated in lots of places in the code). I'm not sure what the best approach is - strict enforcement (i.e. an error rather than a warning) is probably overkill, but we should potentially warn more (e.g. warn when creating an `ImageStack` which has an unsupported data type, and warn in recipe modules which down-cast).

barentine, 2020-12-08:
I opened the stack in PYMEImage and it looked fine.

Not trying to just be a thorn in the side David, you're absolutely right single prec is fine for most anything, but I do think it's somewhat embarrassing if in 2020 we consider a float64 array output an error.

barentine, 2021-02-12:
Maybe more aggravating note on this front that I'm hoping we're OK changing: If a recipe output saves a boolean mask of detected nuclei to h5 (which I hope we can agree is preferable to tif here), one can't `PYMEImage pyme-cluster:///that/file.h5` because we throw hard
```
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/HTTPDataServer.py", line 679, in get_h5_part
    f, length = self._string_to_file(h5f.get_file(part))
  File "/home/ubuntu/code/python-microscopy/PYME/IO/h5File.py", line 123, in get_file
    return self.get_frame(frame_num)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/h5File.py", line 102, in get_frame
    return PZFFormat.dumps((data.squeeze()), compression = self.PZFCompression)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/PZFFormat.py", line 242, in dumps
    raise RuntimeError('Unsupported data type: %s' % data.dtype)
RuntimeError: Unsupported data type: bool

```

barentine, 2021-02-12:
notably you can't download as h5 / tiff through clusterUI of course because that error is on the dataserver

David-Baddeley, 2021-07-12:
Just triaging old stuff here ...

Bool support in PZF (the above error) is a duplicate of #102. The issue downloading from the cluster is that we use PZF as a wire format for the transfer and PZF only supports a few data types. The solution to that issue is to complain more when saving data as anything other than supported types (and/or add more types to PZF).