python-microscopy/python-microscopy #692: [ENHANCEMENT] Make modules searchable in bakeshop `Add Module`

zacsimile, 2020-12-08:
**Is your feature request related to a problem? Please describe.**
It's often difficult to find the module I'm looking for when I open a pipeline and press `Add Module`. There's an increasingly long list of functions (the SKF functions take up a bunch of space), and they're not sorted particularly well (e.g. what is in `localisations` vs. `measurement` is a tossup). I tend to remember the name of the module I want rather than where it's located in the list, and it takes a minute to find the module I want.

**Describe the solution you'd like**
I'd like a search bar where it says `Module name` in the image below that live-filters the list based on what I type.

[image]

**Are you prepared to work on it?**
Yes

David-Baddeley, 2020-12-08:
Great idea (you can already do a bit of filtering by starting to type in the tree view, but it's pretty flaky and might not work on all platforms).

More generally we should probably do a bit of sorting and better document what goes where. The general rule is image data -> measurement, point data > localisations.

zacsimile, 2021-02-10:
Let's add to this "reorganize the recipes". Why is `FilterTable` not under `filters`? Why do we need `tablefilters`?

David-Baddeley, 2021-02-24:
to `filters` and `tablefilters` it's image data vs tabular data - we could definitely change the names to, e.g. `image_filters` and `table_filters` or similar.

I hadn't realised how much of a problem the lack of filtering was on wx4 until recently - it works quite nicely on wx3 without changes, but should be good on wx4 with #886

closed 2021-02-24 by David-Baddeley