python-microscopy/python-microscopy #693: [BUG] recipe rule tasks not being marked as complete?

barentine, 2020-12-08:
**Describe the bug**
Looks like recipe rule tasks are not getting marked as complete. This can be particularly weird when you are queuing acquisition tasks from a recipe rule, and it works, and then it works again, and then it works a third time ;)

**To Reproduce**
Steps to reproduce the behavior:
1. run a recipe rule
2. see the task get assigned
3. see the CPU usage on a node ramp up
4. See the output file save, see the output acquisition queues queue
5. look at the rulenodeserver temrinals and see 'time elapsed' debug statement from the module you ran, nothing bad
6. unpause your acquisition queue and let the scope run wild
7. wait 10 minutes and see the same tasks get queued again, now with a 'timeout' marked on the ruleserver
[image]

**Expected behavior**
mark recipe rule tasks complete when they've executed error free and saved outputs error free

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10 intrument, ubuntu cluster
 - Python version: 3.6
 - numpy version: 1.16 and 1.19
 - wxpython version [if a UI bug]: 4.0.4

**Additional context**
would make sense in my eyes to change the default number of retries for recipe rules to 1. This should also probably be paired with `optimal-chunck-size=1` in the recipe rule default template, as mentioned in #599

barentine, 2020-12-08:
Note the comment about changing default number of retries for a recipe rule would generally mean allowing the number of retries to be specified in a rule template, rather than current code which allows it to be spec'd with a config option on the ruleserver and is global for all rules

David-Baddeley, 2020-12-08:
I'm not so sure there is a case for having a different number of retries for recipes, although there could be one for having a longer timeout. I definitely don't think that we should be specifying the number of retries in the rule template, as this adds a whole new level of complexity where it shouldn't be present (i.e. if something makes this necessary, it's a bug we should fix elsewhere). When submitting a rule, you shouldn't have to think about how many retries you should attempt, but rather just expect it too run.

barentine, 2020-12-08:
That would certainly be nice ;) Having a look at that now

David-Baddeley, 2020-12-08:
There's nothing obvious in `taskWorkerHTTP` which should result in the results for recipe tasks not getting filed. For both this and the remaining issues with accounting etc, it might make sense to set up some task-distribution test code (i.e. stuff which will run with no-op recipe/localisation tasks which we can use to stress test the infrastructure without needing to do it on real data).

barentine, 2021-01-13:
Haven't gotten around to setting up more testing infrastructure, but tinkering with things this morning it appears the issue is not in a lack of putting the failure on the `taskWorkerHTTP.taskWorker.resultsQueue` (hopefully completion too, but only testing failures at the moment) but rather that we don't regularly check that queue if there aren't more tasks to get. In other words you can hit a failure, but if we don't have 50 more tasks queued you'll never know about it. Would be nice to add a timeout somewhere (which would also clean things up fast rather than waiting forever to do the last couple frames of a series if we don't even have more in the queue)

barentine, 2021-01-13:
Yeah, found the issue, introduced in 0b3760d  

We used to pop off queueURLs as we looped through, so we'd check each once and continue, now we just hang on the while and keep waiting until we have tasks

closed 2021-01-20 by barentine