python-microscopy/python-microscopy #694: [BUG] core dumping on binary fill holes with bioformats input

barentine, 2020-12-09:
**Describe the bug**
Looks pretty nasty, and pretty java-y. Will try loading h5's and running the same recipe

<details>

```
$ dh5view Documents/BewersdorfLab/DruggedTudors/Snd1_Both_55.tiff 
INFO:numexpr.utils:NumExpr defaulting to 8 threads.
/Users/Andrew/code/python-microscopy/PYME/DSView/splashScreen.py:117: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
Starting main loop
Loading data
filename == /Users/Andrew/Documents/BewersdorfLab/DruggedTudors/Snd1_Both_55.tiff
Bioformats:loading data
DEBUG:javabridge.jutil:Creating JVM object
DEBUG:javabridge.jutil:Launching VM in non-python thread
DEBUG:javabridge.jutil:Attaching to VM in monitor thread
DEBUG:javabridge.jutil:Signalling caller
10:44:39.992 [Thread-1] DEBUG loci.common.NIOByteBufferProvider - Using mapped byte buffer? false
10:44:40.020 [Thread-1] DEBUG loci.formats.ClassList - Could not find loci.formats.in.URLReader
java.lang.ClassNotFoundException: loci.formats.in.URLReader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_65]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_65]
	at java.lang.Class.forName0(Native Method) ~[na:1.8.0_65]
	at java.lang.Class.forName(Class.java:264) ~[na:1.8.0_65]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.<init>(ClassList.java:138) [bioformats_package.jar:6.5.1]
10:44:40.023 [Thread-1] DEBUG loci.formats.ClassList - Could not find loci.formats.in.SlideBook6Reader
java.lang.ClassNotFoundException: loci.formats.in.SlideBook6Reader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_65]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_65]
	at java.lang.Class.forName0(Native Method) ~[na:1.8.0_65]
	at java.lang.Class.forName(Class.java:264) ~[na:1.8.0_65]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.<init>(ClassList.java:138) [bioformats_package.jar:6.5.1]
10:44:40.023 [Thread-1] DEBUG loci.formats.ClassList - Could not find loci.formats.in.ScreenReader
java.lang.ClassNotFoundException: loci.formats.in.ScreenReader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_65]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_65]
	at java.lang.Class.forName0(Native Method) ~[na:1.8.0_65]
	at java.lang.Class.forName(Class.java:264) ~[na:1.8.0_65]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.<init>(ClassList.java:138) [bioformats_package.jar:6.5.1]
10:44:40.369 [Thread-1] DEBUG loci.formats.ClassList - Could not find loci.formats.in.URLReader
java.lang.ClassNotFoundException: loci.formats.in.URLReader
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424) ~[na:1.8.0_65]
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331) ~[na:1.8.0_65]
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357) ~[na:1.8.0_65]
	at java.lang.Class.forName0(Native Method) [na:1.8.0_65]
	at java.lang.Class.forName(Class.java:264) [na:1.8.0_65]
	at loci.formats.ClassList.parseLine(ClassList.java:196) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.parseFile(ClassList.java:258) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.<init>(ClassList.java:138) [bioformats_package.jar:6.5.1]
	at loci.formats.ClassList.<init>(ClassList.java:122) [bioformats_package.jar:6.5.1]
	at loci.formats.ImageReader.getDefaultReaderClasses(ImageReader.java:80) [bioformats_package.jar:6.5.1]
	at loci.formats.in.FilePatternReader.<init>(FilePatternReader.java:67) [bioformats_package.jar:6.5.1]
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method) [na:1.8.0_65]
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62) [na:1.8.0_65]
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45) [na:1.8.0_65]
	at java.lang.reflect.Constructor.newInstance(Constructor.java:422) [na:1.8.0_65]
[... 378 lines cut ...]
sphinxcontrib-qthelp      1.0.3                      py_0  
sphinxcontrib-serializinghtml 1.1.4                      py_0  
sqlite                    3.31.1               ha441bb4_0  
tempora                   4.0.1              pyhd3eb1b0_0  
tk                        8.6.8                ha441bb4_0  
toolz                     0.11.1                     py_0  
toposort                  1.5                      py37_3    david_baddeley
tornado                   6.1              py37h9ed2024_0  
tqdm                      4.42.1                     py_0  
traits                    6.1.1            py37haf1e3a3_0  
traitsui                  7.1.0                      py_0  
ujson                     2.0.3            py37h0a44026_0  
urllib3                   1.25.11                    py_0  
wheel                     0.36.0             pyhd3eb1b0_0  
wxpython                  4.0.4            py37hebc47a7_0  
xz                        5.2.4                h1de35cc_4  
yaml                      0.1.7                hc338f04_2  
zc.lockfile               2.0                        py_0  
zeroconf                  0.28.0                   py37_3    david_baddeley
zlib                      1.2.11               h1de35cc_3  
zstd                      1.3.7                h5bba6e5_0  

```

</details>

barentine, 2020-12-09:
Note the recipe ran fine (and I opened outputs) without the fillholes, and I did fillholes without 'per frame' checked, radius 10 pixels

zacsimile, 2020-12-09:
Following up:

```
- filters.GaussianFilter:
    inputName: input
    outputName: gauss
    processFramesIndividually: false
    sigmaX: 15.0
    sigmaY: 15.0
    sigmaZ: 9.0
- processing.Threshold:
    inputName: gauss
    method: otsu
    outputName: otsu
    processFramesIndividually: false
- processing.BinaryFillHoles:
    inputName: otsu
    outputName: filled
    processFramesIndividually: false
    radius: 10.0
```

segfaults, as @barentine  pointed out. If I change the `BinaryFillHoles` to have `radius: 3.0`, it works fine. `BinaryErosion` and `BinaryDilation` similarly work.

David-Baddeley, 2020-12-09:
It's not necessarily a java problem - might just be that the JVM is capturing a segfault that is occuring elsewhere.

David-Baddeley, 2020-12-09:
looks like it might be a bug in `scipy.ndimage`