python-microscopy/python-microscopy #695: [BUG] pymeimage recipe tab gets stuck in circular loop depending on order of operations

zacsimile, 2020-12-09:
**Describe the bug**
If I add a more than one module to the `pymeimage` recipe I get a circular reference issue and can't link recipe elements.

[image]
 

Hitting `Apply Text Changes` gives me the error.

```
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/recipeGui.py", line 502, in OnApplyText
    self.recipes.UpdateRecipeText(self.tRecipeText.GetValue())
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/recipeGui.py", line 619, in UpdateRecipeText
    self.activeRecipe.update_from_yaml(s)
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 870, in update_from_yaml
    return self._update_from_module_list(l)
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/base.py", line 823, in _update_from_module_list
    mod.set(**md)
  File "/Users/zachcm/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traits/util/deprecated.py", line 31, in wrapper
    return fn(*args, **kw)
  File "/Users/zachcm/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traits/has_traits.py", line 1376, in set
    trait_change_notify=trait_change_notify, **traits
  File "/Users/zachcm/opt/miniconda3/envs/pyme/lib/python3.7/site-packages/traits/has_traits.py", line 1368, in trait_set
    setattr(self, name, value)
  File "/Users/zachcm/Code/python-microscopy/PYME/recipes/traits.py", line 24, in validate
    raise TraitError('Assigning "%s" to input "%s" would result in a circular reference (value is in module outputs).' % (value, name))
traits.trait_errors.TraitError: Assigning "filtered_image" to input "inputName" would result in a circular reference (value is in module outputs).
```

I also get this error on adding the `Threshold` recipe tile initially. It seems to think I'm creating a circular reference to `filtered_image` even when I properly change input/output names.

**Expected behavior**
I should be able to add recipes and connect them in sequence.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15.7
 - Python version: 3.7.9
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2020-12-09:
Hmm, I can't replicate (or rather, I get an error initially when trying to change it on add, but this is expected, because the output name is "filtered_image" (you need to change this before you can set the input to the same thing). No issues with apply text changes etc ... or after I've added it. 

As far as I can see things are working correctly (preventing you from adding an incorrectly wired module).

zacsimile, 2020-12-09:
Sorry, so I can't use the output name `filtered_image` at all?

David-Baddeley, 2020-12-09:
not if the input is `filtered_image`

David-Baddeley, 2020-12-09:
but otherwise it's fine

zacsimile, 2020-12-09:
But quite clearly in my example the output isn't `filtered_image` and the input is...

David-Baddeley, 2020-12-09:
yeah, that's odd. I don't understand what is happening to make you get stuck - I can't reproduce it.

zacsimile, 2020-12-09:
Ah. I can consistently. Are you on py37?

zacsimile, 2020-12-09:
It works fine for me in `pymevis` but I cannot connect tiles easily in `pymeimage`. Just keep getting this kind of error.

David-Baddeley, 2020-12-09:
Just tried 37 and still seems to work

zacsimile, 2020-12-09:
I think it is `filtered_image`. It appears to be reserved and/or not update when I change it. This works fine:

[image]

David-Baddeley, 2020-12-09:
ahh - it seems that the apply text changes button is not working at all. Not sure why.

David-Baddeley, 2020-12-09:
(it's not generating a new error, that error is left over from when you were putting the recipe together in the first place)

zacsimile, 2020-12-09:
Ah, it seems I can't change `input` to `filtered_image` in the Threshold tile even if I change output to something else (`otsued`) first on adding. But if I change `filtered_image` to `otsued` and `input` to `another_name` while adding the Threshold tile, it works fine.

zacsimile, 2020-12-09:
`filtered_image` can't be an input name if the tile initially has an output name of `filtered_image`, even if I change it.

David-Baddeley, 2020-12-09:
works fine with the graphical editor (i.e. not applying text changes)

zacsimile, 2020-12-09:
`Apply text changes` worked during my testing of #694. I think that button does work, I think the initial add checks if input=output before checking for updated output.

David-Baddeley, 2020-12-09:
actually, that makes sense, it iterates through all the trait validators as it initially sets the traits, as such if you set the input to the default value of the output it complains :/

zacsimile, 2020-12-10:
Since a fair number of tiles have the same output name by default, I think this is fairly easy to run into. Can we change the behavior to check input_name against the updated output_names?

David-Baddeley, 2020-12-10:
definitely a reasonable idea. Implementation is going to be a bit tricky though (potentially involving re-writing the constructor to change the order in which traits are set, which sounds pretty messy).

We might be able to get away with not checking both inputs and outputs for circular references and just rely on the output validation to pick these up, although that would be potentially risky as I don't think we can assume things get set in a deterministic order.

zacsimile, 2021-02-10:
Bumping this because I'm running into it so frequently. This is a tremendously frustrating aspect of using recipes. I'll add that I have to actually type in the data source name to change the input when I get stuck in this loop. The dropdown doesn't work, for whatever reason, it always restores to the default value.

David-Baddeley, 2021-02-24:
should be fixed

closed 2021-02-24 by David-Baddeley