python-microscopy/python-microscopy #697: [BUG] DCAM multiview view-change subarray

barentine, 2020-12-09:
posting quick during meeting - apologies for incompleteness
**Describe the bug**
Invalid subarrays for multiview orca flash which are valid if you go from full chip to those views, but not from other active views
```
DEBUG:PYME.IO.clusterIO:Request for b'http://10.150.7.4:15348/Bergamot/2020_8_13/EMC0103S2/000/' failed with error: 404
DEBUG:PYME.Acquire.Hardware.Camera:setting chip ROI
-128361 128361 -1
(1597343378, 44004) 62824 0
DEBUG:PYME.Acquire.Hardware.HamamatsuDCAM.HamamatsuORCA:Setting ROI: x0 744.0, y0 896.0, w 256.0, h 256.0
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\frameWrangler.py", line 227, in onExpReady
    d = self.getFrame()
DEBUG:PYME.Acquire.Spooler:Disconnecting from frame source
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\frameWrangler.py", line 198, in getFrame
    self.cam.ExtractColor(cs,0)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\Camera.py", line 1038, in ExtractColor
    self.camera_class.ExtractColor(self, self.chip_data, mode)
DEBUG:PYME.Acquire.Spooler:Frame source should be disconnected
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\HamamatsuDCAM\HamamatsuORCA.py", line 300, in ExtractColor
    raise DCAMZeroBufferedException()
PYME.Acquire.Hardware.HamamatsuDCAM.HamamatsuORCA.DCAMZeroBufferedException
DEBUG:PYME.Acquire.ui.rules:htsms-tile
PatchedServiceInfo(type='_pyme-taskdist._tcp.local.', name='PYMERuleServer: smeagol._pyme-taskdist._tcp.local.', address=b'\n\x96\x07\x82', port=15346, weight=0, priority=0, server='PYMERuleServer: smeagol._pyme-taskdist._tcp.local.', properties={}) b'\n\x96\x07\x82'
DEBUG:root:Successfully created recipe rule
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ActionManager.py", line 140, in Tick
    self.isLastTaskDone = fcn(**args)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\SpoolController.py", line 421, in StartSpooling
    self.spooler.StartSpool()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\HTTPSpooler.py", line 229, in StartSpool
    sp.Spooler.StartSpool(self)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Spooler.py", line 133, in StartSpool
    self.protocol.Init(self)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\protocol.py", line 125, in Init
    for e in self.metadataEntries:
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\protocol.py", line 134, in OnFrame
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire/Protocols\htsms-tile.py", line 17, in set_state
    scope.cam.enable_multiview(views)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\Camera.py", line 978, in enable_multiview
    self.SetROI(*self.chip_roi)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\HamamatsuDCAM\HamamatsuORCA.py", line 235, in SetROI
    self.setCamPropValue('SUBARRAY HPOS', x1)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\HamamatsuDCAM\HamamatsuDCAM.py", line 457, in setCamPropValue
    "dcamprop_setvalue")
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\HamamatsuDCAM\HamamatsuDCAM.py", line 500, in checkStatus
    self.getCamInfo(fn_return))
PYME.Acquire.Hardware.HamamatsuDCAM.HamamatsuDCAM.DCAMException: DCAM error for dcamprop_setvalue with return value -2147481557: Invalid Subarray!
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10]
 - Python version:
 - numpy version:
 - wxpython version [if a UI bug]:

**Additional context**
Add any other context about the problem here.

barentine, 2020-12-09:
discussion with David:
try 'setting' ROI, then set a smaller one (debugging the ORCA code)

barentine, 2020-12-09:
shifting subarray out of bounds by moving position first -> turn off subarray mode first, change, then turn on again (or just return if we're setting ROI to full chip)

David-Baddeley, 2020-12-09:
Just returning might without setting a full-size subarray might cause problems with future `GetROI()` calls - probably safer to set a full chip subarray.

closed 2020-12-14 by barentine