python-microscopy/python-microscopy #700: [ENHANCEMENT] Sorted colourmap names

David-Baddeley, 2020-12-10:
Make frequently used colourmap names occur at the top of the drop-down lists to make them a easier to find. General usability enhancement following today's discussion.

**Proposed implementation**

This would also be a chance to address some current matplotlib deprecation warnings and remove our nasty monkey-patching of the matplotlib colourmaps

- make a new file, e.g. `PYME.misc.colourmaps` or PYME.ui.colourmaps`
- use this to proxy `matplotlib.cm`
- include our hacked colourmaps here rather than monkey-patching matplotlib in extraCMaps
- replace all matplotlib.cm references with our local file
- sort `.cmapnames` as we see fit

###Open Questions:

- Try and find a logical order (e.g. r,g,b,c,m,y, grey, R,G,B,C,M,Y, .... all the rest)?
- or remember most recently used?
- or some combination, e.g. 'Bayesian' ordering - in which we attempt to find a logical ordering and then modify it if based on user history.
- if remembering most recently used, should this be per-session, or persist across sessions?

zacsimile, 2020-12-11:
- A logical order would be nice, though not strictly necessary
- Remembering the most recently used would be the most helpful thing
- Bayesean ordering could be useful for color selection in multiple channels
- I think history should persist across sessions
- It would also be nice to save the most recent point type selection, alpha, and point size (e.g. I prefer pointsprites, alpha 0.04 and size=30, and would be very happy to have all images open that way by default)

David-Baddeley, 2021-12-05:
#700 has brought up a few more questions re. any smart ordering:

1) this should probably be specific to data type (e.g. images, points, surfaces)
2)  (1) means we'd need to convey intent when accessing the colourmaps if we are going to log / record most frequently used (or record most frequently used locally, which would lead to lots of duplication).