python-microscopy/python-microscopy #701: [ENHANCEMENT] Display # localizations in h5r files on clusterUI

barentine, 2020-12-11:
**Is your feature request related to a problem? Please describe.**
scope users want a quick ballpark of whether things are going OK. Displaying length of tabular types would be a stop gap until we have #347 or similar

**Describe the solution you'd like**
this, but with #localizations for h5r/hdf files instead of #frames
[image]

**Describe alternatives you've considered**
really would be nice to just run a PYMEVis --server and as long as 'live view localization' box is checked have it update the HTTPDatasource every time we push a localization rule

**Additional context**
I realize this means opening the file, but that isn't different from cluster h5's.

**Are you prepared to work on it?**
way more interested in the alternative, which I mostly already did

David-Baddeley, 2020-12-11:
More or less on the TODO list, with integration of the sample database (would also give us image thumbnails and let us see some of the metadata). A few potential stumbling blocks:

- really needs to have results cached to give acceptable performance (.h5 listing on cluster of one is a bit of an abberation which we get away with because of the much lower file numbers and performance demands in cluster of one use cases).
- to do the above, we need to be able to detect that a series is finished / in progress

A potential alternative would be to just display file sizes, which tend to be a very good proxy for the number of localisations and are significantly cheaper to calculate.

barentine, 2020-12-11:
file size would be a great proxy. Especially since #localizations is just a proxy for a good image anyway, I think filesize would be sufficient here and as you mention something more substantial as a follow up (QC html reports generated on click? etc.)

barentine, 2021-01-05:
notably the h5 # files is comically whack at the moment
[image]

[image]