python-microscopy/python-microscopy #708: [BUG] fresh install, `PYMEClusterOfOne --clusterUI` silently fails clusterUI if django isn't manually installed

barentine, 2020-12-15:
**Describe the bug**
django is an optional dependency (I dissent, but not a big deal), `PYMEClusterOfOne --clusterUI` fails silently if one doesn't have django installed

**To Reproduce**
Steps to reproduce the behavior:
1. install PYME
2. `PYMEClusterOfOne --clusterUI
3. browser redirects to page not found
4. try running clusterUI standalone and see modulenotfound Django error

**Expected behavior**
warn people they forgot to install django

barentine, 2020-12-15:
might also become relevant for packaging, particularly related to #657, but already an issue on 3.7:
```
$ conda install django=1.11
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                 

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - django=1.11 -> python[version='>=2.7,<2.8.0a0|>=3.5,<3.6.0a0|>=3.6,<3.7.0a0']

Your python: python=3.7

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

```

barentine, 2020-12-15:
note that pip installing any old django 1.11 won't work, but after some convincing (supposed to use django 2.x for py37) they backported a fix https://github.com/django/django/pull/10654

David-Baddeley, 2020-12-15:
The above mentioned packaging / conda issue is one of the reasons *why* django is an optional dependency.

- django is generally not backwards compatible between versions (i.e. we need to either pin or spend a lot of time constantly re-writing the bits of the code which use django so they work with the most recent version).
- such pinning is highly likely to result in un-resolvable environments.

For the main issue, I agree that it would be nice to warn people if they don't have django installed.

David-Baddeley, 2020-12-15:
We should probably try and see if the code runs on a more recent django and update pins if needed

barentine, 2020-12-15:
1.11 is the last py27 support. We discussed and went to django==1.11.29 in the pip requirements. For the issue I mention here we only need >=1.11.17, but I'm not sure what conda packages are available there for 3.7 (didn't look).

When we drop 1.x we can bump either to 2.2 which will EOL mid 2022, or if we wait til maybe March we can try jumping straight to 3.2 which will be another LTS https://www.djangoproject.com/download/

David-Baddeley, 2020-12-15:
What I meant is have we actually tried the current code on django 2.x? Had a quick look at the breaking changes and I suspect it might actually run without modification.

barentine, 2020-12-16:
I have not tried django 2.x. Will try and remember to give that a go on my laptop sometime

barentine, 2020-12-16:
I guess I wonder why we direct clusterUI stderr to file in cluster of one. I can see redirecting stdout for any django app, but errors would be helpful and then we would get the 'module not found' reporting for free here

David-Baddeley, 2020-12-18:
Django gets supper chatty and will drown out any other error messages (which is why we redirect to file). Not sure how much of this goes to stdout vs stderr, but was easiest to just redirect both than try to quantify. The logging module documentation suggests that log messages go to stderr rather than stdout (i.e. this is the one we need to suppress).

David-Baddeley, 2020-12-21:
Looks like it will be fairly simple to make things work on 1.11.0 <= django < 2.2.0. (i.e. add support for 2.0 and 2.1 without effecting 1.11)

David-Baddeley, 2020-12-23:
Might be able to close with merge of #730 ...

closed 2020-12-27 by barentine