python-microscopy/python-microscopy #709: [BUG] cluster h5's don't load in very conveniently for recipes

barentine, 2020-12-15:
**Describe the bug**
Using the `recipe_form` template/view
```
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 340, in computeLoop
    recipe.execute()
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 664, in execute
    m.check_inputs(self.namespace)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/base.py", line 137, in check_inputs
    raise MissingInputError('Input "%s" is missing from namespace; keys: %s' % (input, keys))
PYME.recipes.base.MissingInputError: Input "input" is missing from namespace; keys: ['PZFImageData', 'PZFImageIndex']
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/IO/h5rFile.py", line 249, in _pollQueues
    self._h5file.flush()
  File "/opt/miniconda3/lib/python3.7/site-packages/tables/file.py", line 2696, in flush
    self._check_open()
  File "/opt/miniconda3/lib/python3.7/site-packages/tables/file.py", line 2150, in _check_open
    raise ClosedFileError("the file object is closed")
tables.exceptions.ClosedFileError: the file object is closed
```

**To Reproduce**
Steps to reproduce the behavior:
1. run `PYMEClusterOfOne --clusterUI`
2. navigate to a directory with an h5 series
3. note that we're currently getting the 'file' count very wrong (at least on later pytables/whatever)
4. select an h5 series
5. click `run recipe` on the lhs
6. load a recipe
7. run
8. pop open your RuleNodeServer terminal to see the above traceback

**Expected behavior**
in this instance, `PZFImageData` would've been nice to be 'input'

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: osx 10.15
 - Python version: 3.7
 - numpy version: 1.19
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2020-12-15:
We should probably handle .h5 as an ImageStack rather than using the generic code (see line 1101 of recipes.base). This is potentially particularly important when we're doing PZF in HDF (as is the default cluster of one case).

I'd try removing .h5 from the list of extensions on 1101 and seeing how that works.

barentine, 2020-12-16:
should be set, will reopen if not

closed 2020-12-16 by barentine