python-microscopy/python-microscopy #711: [BUG] Weird case of ROI not clipping correctly but giving no error

lukasfue, 2020-12-15:
**Describe the bug**
Weird case of ROI clipping not working. Might be hard to reproduce, but thought it was worth documenting at least.

**To Reproduce**
Steps to reproduce the behavior:
DId the following:
1. generated jittered triangulation image 
2. Thresholded with isodata(logbins)
3. Labeled with 100 min size
4. Get ObjectIDs from image in Visgui
5. Clipped to ROI on this labeled data
6. Generated JIttered triangulation image on this ROI
7. Thresholded using isodata(logbins)
8. Labeled with 10 min size
9. Get ObjectIDs from this labeled image with previously labeled points as selected datasource
10. Displayed these labeled objectIDs 
11. Changed selected datasource to previously labeled data and back
12. I tried to crop the second set of labeled objects and it didn't work but gave no error.

**Expected behavior**
ROI to crop

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10
 - Python version: 3
 - numpy version:
 - wxpython version [if a UI bug]:

David-Baddeley, 2020-12-15:
ROI clipping only applies to the `output` data source. Other, intermediate data sources (such as `labels_from_img_8` as you have selected in layer 3) do not have the ROI applied. This is not super well documented or intuitive, but is a result of the the fact that ROI clipping is implemented *after* the processing pipeline.

The best explanation at present would probably be the figure in 
https://python-microscopy.org/doc/ExtendingVisGUI.html

When you set the layer data to anything other than output, you are 'peeking' back in the pipeline to a point before ROI clipping is performed. This will hopefully get a bit cleaner once we combine view clipping and ROIs, rather than having them as two separate functions..

David-Baddeley, 2021-01-28:
removing bug label, but leaving open pending an ROI discussion issue.