python-microscopy/python-microscopy #712: [ENHANCEMENT] Easier camera map acquisition and processing - ROI indexing conventions?

csoeller, 2020-12-15:
**Is your feature request related to a problem? Please describe.**

I am hijacking this issue as a discussion thread for camera maps. On the positive side I have been able to use the very convenient action queuing  interface to make some code to take camera maps across the whole chip even for pretty fast acquisitions. See example screenshots below.

The potential issue I noticed and was going to get some feedback on are subtle pixel indexing issues. I seem to recall that on the Zylas there was a time when ROI indexing started with pixel (1,1) in the origin. I have some older code that seems to strongly suggest that.

Now I get the impression that ROI indexing starts at (0,0). Also when you request a ROI [x0,y0,x1,y1] then this appears to translate into a region `reg[x0:x1,y0:y1]` in numpy parlance, i.e. excluding pixel coordinate x1,y1 from the region selected.

I am a little concerned if there was indeed a change if map registration has been carefully checked (against sufferiung from off by 1 pixel issue or similar). I can envisage tests (and I did test it back in the day) but am now a little uncertain if a convention change occured and if the map generation code has been checked if it is still correct?

**Describe the solution you'd like**
Just clarity if I understand correctly and if there could even be camera to camera differences.

**Additional context**

I show here the preliminary interface for map recording that has worked pretty well for me. I have conducted quick map generation for each sequence with bakeshop + recipe and a final notebnook to combine (for now). This is where the ROI indexing started to worry me.

[image]

[image]

[image]

**Are you prepared to work on it?**
Absolutely.

barentine, 2020-12-15:
Hi @csoeller , the camera map calibration code and everything in remfitbuf is zero-indexed (and using slices). The old 1-index metadata keys are `Camera.ROIPosX` and `Camera.ROIPosY` while the new zero-indexed keys are `Camera.ROIOriginX`/`Y`. These are more checked for (and the 1-indexing accounted for if using ROIPos) in `PYME.IO.MetaDataHandler.get_camera_roi_origin`. Good check on the sCMOS maps! I changed them over in/around [this commit](https://github.com/python-microscopy/python-microscopy/commit/3028d7b87a4dc0db78d31d6eff078250e7e44ef0)

If your Zyla uses 1-indexing and ROIPos metadata you should be fine, but I think ROIOrigin might be slightly preferable (either way, use `get_camera_roi_origin` and you'll get zero-indexed out).

csoeller, 2020-12-15:
Excellent, thanks @barentine for the overview and piointers, just what I was after. This all sounds very reassuring and I will use that info to get the map combination code to work properly. For now in a quick notebook, longer term probably with a suitable recipe.

barentine, 2020-12-15:
Glad that helps, and thank you for working towards knocking off some of the frustration of camera map generation! We do already have a module in this direction in `PYME.recipes.processing.DarkAndVarianceMap`. Short of full documentation, but  if you wanted to chain that, in a recipe called `map-gen.yaml` to run automatically following an acquisition protocol `zyla-map-gen.py`, you could add the following to your init script:

```
    from PYME.Acquire.ui.rules import ChainedAnalysisPanel
    from PYME.cluster.rules import RecipeRuleFactory
    import yaml
    import os

    # add some default pairings
    defaults = {}
    rec_dir = 'C:\\Users\\Bergamot\\PYMEData\\recipes'

    rec = os.path.join(rec_dir, 'map-gen.yaml')
    with open(rec) as f:
        rec = f.read()
    defaults['zyla-map-gen'] = [RecipeRuleFactory(recipe=rec)]

    ChainedAnalysisPanel.plug(main_frame, scope, defaults)
```

then in the panel on the rhs of PYMEAcquire select run on 'spool stop' and run `PYMEClusterOfOne`.

GUI's a bit broken, working on changing it, but if you are really on a roll with this I thought I'd point you to where the alpha stuff lives. At the moment that would just put the output in a subdirectory of the spooled seires. You should be able to stuff the output_dir you want for camera maps into the RecipeRuleFactory init

csoeller, 2020-12-15:
Cool, I might try that rule chaining over the Xmas break as something more fun to do.

So far I have been using the `DarkAndVarianceMap` module via bakeshop and it does its job admirably. So once I figure out that rule factory business definitely the way to go. (I believe I will have to have a Cluster[ofOne] going to make this work?)

I noticed one funny thing though using the `DarkAndVarianceMap` module. The metadata that was in the form of tuples/arrays seems to get clobbered into a long string in the resulting maps, as if the pickling had taken over or something? See screenshot below, source metadata on the left, one of the resulting maps on the right.

[image]

barentine, 2020-12-15:
Glad it's (mostly) working. gen_sCMOS_maps touches Camera.ROI if you're inserting into full maps, and then it uses a tuple, whereas it looks like your camera.GetROI is returning a list (fine, docs don't spec it but I think that's typical)

Do you still hit this if you save the map as h5 instead of tif?

barentine, 2020-12-15:
binning also getting messed up would imply the xmlmdhandler can't handle tuples

csoeller, 2020-12-15:
Quick test (running the recipe in dh5view, then saving the generated darkmap) shows that clobbering occurs with saving to OME Tiff (all array-like metadata) and loading from that stored TIFF but all is fine when saving to h5. I have a feeling this used to work fine. Recent TIFF saving changes?

barentine, 2020-12-15:
Had a quick look and think I see what's up @csoeller. Posted in #713. If we're set here on camera map indexing maybe we close this for now and use the tif metadata ticket for that?

That said if you can think of somewhere it'd help to put more camera map indexing doc I'm sure that would be a welcome PR . @David-Baddeley maybe we should update the https://python-microscopy.org/doc/metadata.html docs to use `Camera.ROIOrigin`?

David-Baddeley, 2020-12-15:
The only recent changes to the tiff handling would be py3 vs py2. I wonder if this is the cause. In general, anything which is not a simple type (int, float, string) gets saved into the XML as a base64 encoded pickle, but should be transparently unpickled.

csoeller, 2020-12-15:
Thanks @barentine, will refer all further tiff related stuff to #713.

Re map/camera chip handling I am now all clear on ROI origin conventions.

The only other point relating to our camera conventions I would like to be sure about is this: is asking for a ROI of shape (x0,y0,x1,y1) by convention **for all cameras** a ROI starting at x0,y0 (including that point and indexing starting at 0) and going up to but excluding x1,y1? That seems to be the Zyla/SDK3 implementation but I was wondering if this is true regardless of camera type (i.e. enforced by all PYME backends).

David-Baddeley, 2020-12-15:
I think I changed all cameras to use numpy style (0 indexed) ROIs but there is a small chance that I missed one. Indexing *should* now be 0-indexed, so if there is still a camera which is 1 indexed that's a bug.

csoeller, 2023-12-09:
Should be ok to close.

closed 2023-12-09 by csoeller