python-microscopy/python-microscopy #713: [BUG] XMLMDHandler pickle->str on setEntry gets caught by ==str on getEntry and no unpickled

barentine, 2020-12-15:
**Describe the bug**
in setEntry things like tuples get pickled and cast to str. In getEntry those same entries are then assumed to be str and not unpickled

**To Reproduce**
see conversation in #712 

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
will leave for @csoeller to fill in, I just looked at the code

**additional context**
tif i/o - our favorite!

David-Baddeley, 2020-12-15:
I wonder if this is py3 specific - the loader uses the `'class'` attribute of the XML node to decide how to treat things, which should be `'pickle'` for anything we've pickled.

barentine, 2020-12-15:
sorry I'd looked quick and missed that David

csoeller, 2020-12-15:
I can confirm that my original observation was py3 based, I just ran a check with py27 based PYME and everything is fine with that. So purely py3 related it seems...

David-Baddeley, 2020-12-23:
This is super odd - I can't reproduce within py3. I wonder if we run into the issue exclusively when trying to load data that was saved under py2 with py3 (suggesting a pickle protocol version issue).

David-Baddeley, 2020-12-23:
Definitely a py2 vs py3 pickle incompatibility problem ... might be solvable

csoeller, 2021-03-31:
After closing #974 by addressing it with #979 I went back to checking what is happening in this bug. I now confirmed that the issue seen here (XMLMDHandler pickle->str on setEntry gets caught by ==str on getEntry and not unpickled) only occurs when py3 generated pickled metadata is opened with a py2 based PYME.

So this is strictly speaking not a backwards compatibility issue at all, rather an issue that deprecated software does not do the right thing in some cases.

Given that scenario, should we just close this bug (and possibly add a warning when unpacking meatadata under py3)?

David-Baddeley, 2021-07-12:
It sounds like this should be able to be closed

closed 2021-07-12 by David-Baddeley