python-microscopy/python-microscopy #714: [BUG] PYME acquisitions lacking a z-controller try to access it nevertheless

csoeller, 2020-12-15:
**Describe the bug**
I regularly run PYMEAcquire in simple configs that lack a z-controller. This inludes certain scopes but also when running init file that only registers the camera, e.g. for camera calibration etc.

During acquisitions there are attempts to write stack metadata which fail as a z-control is not available.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `PYMEAcquire` with simple init file that lacks z-controller (e.g. no piezo)
2. Start acquisition.
3. Non-fatal but nevertheless potentially confusing error is produced.
4. Error shown below:

```
(10.16080325929549, 'spoolController.StartSpooling', {'maxFrames': 500, 'stack': False}, 1609032592.9298954, 1.7976931348623157e+308), spoolController.StartSpooling
DEBUG:PYME.Acquire.SpoolController:Looking for C:\Users\cs463\PYMEData\cs463\2020_12_15\15_12_series_P.h5 on local machine

ERROR:PYME.Acquire.stackSettings:Error writing stack metadata
Traceback (most recent call last):
  File "c:\python-support\pyme-py3\PYME\Acquire\stackSettings.py", line 161, in ProvideStackMetadata
    mdh.setEntry('StackSettings.StartPos', self.GetStartPos())
  File "c:\python-support\pyme-py3\PYME\Acquire\stackSettings.py", line 93, in GetStartPos
    return self._CurPos() - (self.GetStepSize()*(self.GetSeqLength() - 1)*self.GetDirection()/2)
  File "c:\python-support\pyme-py3\PYME\Acquire\stackSettings.py", line 132, in _CurPos
    return self.scope.GetPos()[self.GetScanChannel()]
KeyError: 'z'
DEBUG:PYME.Acquire.Spooler:Disconnecting from frame source

```

**Expected behavior**
Gracefully bypass stack settings when no z-control available.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] all platforms
 - Python version: 2.7, 3.7
 - numpy version: various
 - wxpython version [if a UI bug]: N/A

barentine, 2020-12-16:
Hi @csoeller , which panel did you use in the gui? We have some discussions about how we set up the acquisition panels (proposed ideal being skinning to say which ones you want), but at the moment one of the constants is the simple stack panel, 'seqpanel', which gets added if there are _any_ positioning keys (assumption I guess being that 'z' would be the only one if any?)
[image]
In the seqpanel init we append the stacksettings metadata as a start metadata provider, which is leading to the error you're hitting.

Presumably then the HDFSpoolFrame ('time/blinking series') would be the one you want, and we should modify the check on positioning keys in acquiremainframe.py to specifically look for 'z' before adding the seqpanel

csoeller, 2020-12-16:
Hi @barentine, thanks for looking into it. I don't think I am requesting anything via the init file, in fact it is pretty minimal apart from the camera stuff. Rather than posting the whole thing here, please have a look at [init_Zyla_n.py](https://github.com/csoeller/PYME-exeter-siteconfig/blob/master/init_scripts/generic/init_Zyla_n.py) which is in our siteconfig github repo.

Happy to learn I did something in the file that I shouldn't have.

David-Baddeley, 2020-12-16:
The z settings stuff is saved by default in the base metadata collection, regardless of how you acquire a series and whether you have a piezo or not. We should probably change that.

barentine, 2020-12-16:
Yep, sorry I missed that @csoeller but the seqpanel is also a part of HDFSpoolFrame, so you get it no matter what.

David-Baddeley, 2020-12-18:
I think there are a few points here:

- At present, the stack settings are added in a round-about way, via their GUI controls. This is clearly not desirable.
- Ideally stack settings would be added independently of UI, if and only if there is hardware for doing a stack.
- UI would check for presence of stack settings before adding the relevant controls.

David-Baddeley, 2020-12-21:
#733 should hopefully address the immediate problem, leaving the slightly confusing behaviour that the stack settings get saved even for non-blinking z series when a piezo is present.

closed 2021-01-12 by David-Baddeley